cheapjack/buildyourown #2: Get Ultimakers ordered and get a sweet deal!

cheapjack, 2015-04-08:
@defproc knows them maybe press them in person at Maker Faire?

cheapjack, 2015-04-19:
Makes sense now to seal the deal directly in person on the stand at Maker Faire esp. as we will have a raptor hand with us to show. And if it goes badly we can always use the nerf guns on them. Trigger pulled by Raptor hand, obviously

cheapjack, 2015-04-27:
@DefProc met with Ultimaker people at MakerFaire (imbetween running his insanely popular MFUKLC Nerf gun rifle range) and they were up for lending an additional printer over course of FACT show and got a deal with purchasing 2 of the printers. Result! so we need to follow up and if we can get them sooner rather than later we can calibrate work out bugs at DoES etc prior to install.

LesleyTaker, 2015-04-27:
This is great news guys!! Let me know if I can help with this is any way.
Well done @DefProc!

-- 

Lesley Taker
Curatorial Co-ordinator
FACT (Foundation for Art and Creative Technology)
88 Wood Street
Liverpool, L1 4DQ

Please note, I am only in office until 3pm on a Friday.
tw: @lexy_taylor
t: + 44 (0)151 707 4419
f: + 44 (0)151 707 4445

http://www.fact.co.uk
Admin: +44 (0)151 707 4444
Bookings: +44 (0)871 902 5737

*FACT - Bringing people, art and *
*technology together since 2003*

--------------------------------------------------------------
--------------------------------------------------------------

*Group Therapy: Mental Distress in a Digital Age5 March – 17 May*

Originating from FACT’s extensive work within

mental health and wellbeing, the exhibition

explores the complex relationship between

technology, society, and mental health.

--------------------------------------------------------------
--------------------------------------------------------------

amcewen, 2015-05-02:
@DefProc can you own getting the printers ordered, etc. as you spoke to Alex at Ultimaker GB the most, and dragging whoever in (probably @LesleyTaker for arranging the payment side of things?) as needed?

LesleyTaker, 2015-05-05:
Fine with me- love paying for things!!! ;)

-- 

Lesley Taker
Curatorial Co-ordinator
FACT (Foundation for Art and Creative Technology)
88 Wood Street
Liverpool, L1 4DQ

Please note, I am only in office until 3pm on a Friday.
tw: @lexy_taylor
t: + 44 (0)151 707 4419
f: + 44 (0)151 707 4445

http://www.fact.co.uk
Admin: +44 (0)151 707 4444
Bookings: +44 (0)871 902 5737

*FACT - Bringing people, art and *
*technology together since 2003*

--------------------------------------------------------------
--------------------------------------------------------------

*Group Therapy: Mental Distress in a Digital Age5 March – 17 May*

Originating from FACT’s extensive work within

mental health and wellbeing, the exhibition

explores the complex relationship between

technology, society, and mental health.

--------------------------------------------------------------
--------------------------------------------------------------

DefProc, 2015-05-09:
Email sent. You should have a copy too @LesleyTaker

LesleyTaker, 2015-05-11:
I do! Thanks!

-- 

Lesley Taker
Curatorial Co-ordinator
FACT (Foundation for Art and Creative Technology)
88 Wood Street
Liverpool, L1 4DQ

Please note, I am only in office until 3pm on a Friday.
tw: @lexy_taylor
t: + 44 (0)151 707 4419
f: + 44 (0)151 707 4445

http://www.fact.co.uk
Admin: +44 (0)151 707 4444
Bookings: +44 (0)871 902 5737

*FACT - Bringing people, art and *
*technology together since 2003*

--------------------------------------------------------------
--------------------------------------------------------------

*Group Therapy: Mental Distress in a Digital Age5 March – 17 May*

Originating from FACT’s extensive work within

mental health and wellbeing, the exhibition

explores the complex relationship between

technology, society, and mental health.

--------------------------------------------------------------
--------------------------------------------------------------

cheapjack, 2015-05-26:
Currently awaiting delivery of 2 Ultimakers at FACT (missed chance to get delivered to DoES) current status at 11:30 26/05/2015 
Not sure of status of loan printer

Consignment Number 	0484212
Despatch Date 	22 May 2015
Number of Parcels/ Items 	2
Total Weight 	32
Consignment Service Code 	ND16
Sender's Reference 	MGC REF: 02138/UK
Delivery Summary
Tracking Summary
Consignment received at Sending Depot	YES
Consignment received at APC Hub	YES
Consignment received at Delivery Depot	YES
Consignment delivered	NO

LesleyTaker, 2015-05-26:
These are at FACT! Just arrived now.

DefProc, 2015-05-26:
One is unboxed and printing it's test print. All looking great so far. I'll leave the other for @cheapjack to have a go with from scratch.

closed 2015-05-28 by DefProc