cheapjack/buildyourown #4: Contact the new designer on data and brief

cheapjack, 2015-04-08:


cheapjack, 2015-04-13:
Notes from Lesley on new data designer: ***Although Stefanie is unable to do it (I saw Alma emailed you) we are still employing someone to do this and need the basic information which we are going to ask them to visualise so they know how much work is involved etc. Alma suggested a map might be a good idea to show the reach of the project and she has a couple of great designers she is approaching, including http://niceandserious.com/ who are top of the list right now. Again, would be great to get put directly in touch with the charity about this, and if you have any ideas about what you would specifically like to be in there, for you to let us know. ***

almataki, 2015-04-13:
This is the map that Lesley refers to: http://www.zeemaps.com/pub?group=609826&legend=1&geosearch=1&search=1&locate=1&list=1&shuttered=1&add=1 I will let you when I manage to speak with nice and serious and you can discuss this further with them. It seems to me that an "attractive" version of this map is pretty straightforward given the limited time we have but also quite comprehensive of the project's global reach. we don't want this to take too much of your time, this is why Lesley also suggested we speak directly to e-NABLE in case there is other existing data we could use.

amcewen, 2015-04-13:
Once issue #14 is done, we can feed that info through to the designer

almataki, 2015-04-24:
Hi guys, I just met with the curators and it looks like, given the time and resources that getting an illustrator to create a big beautiful version of the map above might be the best course of action at this point. A large approx. 2m wide print of an illustration of this map will still convey the global impact of the project dont you think?

almataki, 2015-04-24:
I should have also mentioned that 'niceandserious' aren't available to work on this so the curators will suggest some other illustrators.

almataki, 2015-05-19:
Hi guys a quick update on this. I'm finally working on this with Rada Lewis who is developign a static version of the map I've mentioned before. here is a snapshot of a small part of it of how it's shaping up
[image]

almataki, 2015-05-19:
do you know what OST means from the different types of contributors?

almataki, 2015-05-19:
The full print will be 2m wide x 85cm. Hope you like it I think it will be impressive!

cheapjack, 2015-05-20:
Looks awesome, sorry we've not had chance to input on this but Rada's work is looking awesome. Youc an find a key to all the terms here http://enablingthefuture.org/2014/03/21/make-your-mark/
OST seems to refer to an Organizational Support Team ie a volunteer directly associated with e-NABLE i think. :)

almataki, 2015-05-20:
phew! I was worried you may not like it and it is all happening so fast that there wasn't really much time for feedback. I will send you the final version before printing but glad you like the aesthetic. Thanks for the info!

amcewen, 2015-05-23:
That looks great!  I guess it also means we can close this issue now :-)  Win-win! ;-)

closed 2015-05-23 by amcewen