cheapjack/buildyourown #6: Give Lesley Taker some dates for a Brochure deadline on Monday 13th April

cheapjack, 2015-04-10:
I've not had much time to work on this, so not been able to secure any events yet. WIll do my best to set a date for a talk with DaDaFest on Monday if I can.

I've setup a ***calendar*** here now and invited Lesley, @amcewen @DefProc to manage 
http://bit.ly/prostheticscal

I've put in 2 saturday workshops initially after the start of the show and avoided makerday and LIverpool make fest but assumed a saturday after the event  could be a good time to do something so I could be there for that as I can see theres a beginners arduino booked for that saturday at DoES.

Ive called the saturday 20th one where we explore modifications of the raptor hand so more of a raptor hack. I will write some copy for that here or in the main repo

If we can use that calendar to point to events that may be the best we can do for now.

amcewen, 2015-04-13:
The initial dates look good to me.  From the initial budget sheet it seems we've got another two to schedule in.

Presumably an assembly day would be useful as the third workshop, maybe on 18th of July?  And then 22nd August for the final one?

cheapjack, 2015-04-15:
Adrian and I have just suggested some dates that work for our availability, make sense to project and consider some of Amy's & @LesleyTaker 's date suggestions. We will add copy for them by the end of the week (Fri 17th April)

closed 2015-04-15 by cheapjack

LesleyTaker, 2015-04-16:
Hey Ross,
Just linking in Amy here so she can check it doesn't clash with anything.
She's leading on all events programme for BYO. Amy, suggestions below.
Ta!
L

FACT Liverpool events:

   - Measuring Up *Saturday 6th June*
   - What's your superpower? A Prosthetic Hack!
   <https://github.com/cheapjack/buildyourown/blob/master/events/prostheticshack>
    *Sunday 20th June*
   - Assembly workshop *Saturday 18th July* & *Saturday 15th August*
   - TBD

-- 

Lesley Taker
Curatorial Co-ordinator
FACT (Foundation for Art and Creative Technology)
88 Wood Street
Liverpool, L1 4DQ

Please note, I am only in office until 3pm on a Friday.
tw: @lexy_taylor
t: + 44 (0)151 707 4419
f: + 44 (0)151 707 4445

http://www.fact.co.uk
Admin: +44 (0)151 707 4444
Bookings: +44 (0)871 902 5737

*FACT - Bringing people, art and *
*technology together since 2003*

--------------------------------------------------------------
--------------------------------------------------------------

*Group Therapy: Mental Distress in a Digital Age5 March – 17 May*

Originating from FACT’s extensive work within

mental health and wellbeing, the exhibition

explores the complex relationship between

technology, society, and mental health.

--------------------------------------------------------------
--------------------------------------------------------------