cheapjack/buildyourown #11: Update copy to include wider connections to reach and Nottingham hackspace

cheapjack, 2015-04-13:


amcewen, 2015-04-13:
I'd already made a start on this at https://github.com/cheapjack/buildyourown/blob/master/Background.md

marionofnorwich, 2015-07-17:
I can let ypu have copy for Norwich Hackspace but is there a more recent version somewhere?

almataki, 2015-07-23:
@marionofnorwich Can you please send me a short 'about' blurb for Norwich Hackspace (similar to DoES's) to include in the revised exhibition guide for Norwich and I can send you the new copy of the exhibition panel through next week.

marionofnorwich, 2015-07-27:
Hello Alma, yes I will. Sorry this got caught wrongside a I picked it up
via mobile
Marion

Marion Catlin
3 Trowse House Cottages
Bracondale Millgate
Norwich NR1 2EQ

Telephone 01603 765353
Mobile 07946 261651

cheapjack, 2015-10-26:
I think this can be closed now. 

I passed on our participants names to Melina Brown at e-NABLE, her contact below, to make sure they were in the e-nable mailing list and get any support available, may be worth doing this at the end of the Norwich run

Enable Matcher <enablematcher@gmail.com>

closed 2015-10-26 by cheapjack