cheapjack/buildyourown #12: Get all exhibition presentation completely cemented with the Bloors this week

cheapjack, 2015-04-13:
Be great to get all exhibition presentation completely cemented with the Bloors and us this week so they can order all materials (they're getting worried about import dates etc). I am also going to send you the draft of the exhibition credits etc today, and would be great if I can get those back by the end of the week with any additions or amends you need to make.

amcewen, 2015-04-13:
Is the best way forward with this a conf call, or does it make more sense to spec out what will be on each plinth, etc. in more detail and use that to drive the requirements?

(Thinking out loud here - might need both for it to be a productive conf call...?)

cheapjack, 2015-04-15:
Hi I think we are happy to raise issues and tasks toward the exhibition design and Simon & Tom are responding to issues on there (thanks guys!) now. I've updated the [main exhibition setup document](https://github.com/cheapjack/buildyourown/blob/master/Exhibition/FACT/FACTSpaceSetup.md) to be a bit clearer on numbers of pegboard panel sections(4) to number of hit and miss panels(5).

I dont think we need a conference call or anything now. Just comment or raise an issue if you have any things that need resolving guys @bloors :+1:

cheapjack, 2015-04-27:
The Bloors updated ideas on space setup last week which we need to comment on. This is the first option embedded below that look the best but unsure on the module (prob for a text panel) that splits up the space. Its a decent sized space we can add tables/seating if required to accomodate Amy's concern on numbers on the Measuring workshop.

Ive updated the exhibition setup file to include

[image]
[image]

cheapjack, 2015-04-27:
Need to comment on "We're looking at options for the story plinth, we've got three at the moment, one's a long linear one along a wall, the second comes out from the wall (fixed) at 90 degrees but is more like a table with objects viewed from two sides, and the third is a free stranding table. there is room for wheelchair access all the way around, we're thinking the height will be 700mm, about the height of a table, but it could be up to 800mm high, I'd say any lower than 700mm is getting a bit too low for viewing objects for most standing adults. 
We're imagining the objects will be on the 'sub plinth' panels on the table top with space for any interpretation texts next to each block. The size (and number) of these is flexible but we need to work out the overall dimensions so we can get the bigger MDF sheets and the metalwork all pre-cut. the scaffold fixings will be different depending on the arrangement so this needs working out soon. 

Also if any of the objects need to be in vitrines the dimensions need confirming soon so these can be ordered.

Do you have preference to the layouts below? the first long 'bench' type version does leave a lot of empty room in the space which the Fact gang didn't think worked so well, but it's more like the previous designs we've discussed.

Cheers
Tom"
*Option 1*
[image]
[image]
*Option 2*
[image]
[image]
*Option 1*
[image]
[image]

almataki, 2015-04-27:
I think conceptually options 2 and 3 are much more interesting in the space. being able to walk around the storyplinth rather than sticking them into a line by the wall.

cheapjack, 2015-04-27:
Ok I think option 3 works better then: you can get round the back of the story table with that one then and still find space for extra people... I'm assuming all the green area is for us, the screen in the middle of the space seems superfluous and splits the story from the production bit on the other side so not sure it's a good idea? Also I cant quite see the orientation; ie where the G1 window is?

LesleyTaker, 2015-04-27:
The G1 window is closest to the table with seats on, where the printers
are: that runs all along the wall which also runs along the ramp into G1.

-- 

Lesley Taker
Curatorial Co-ordinator
FACT (Foundation for Art and Creative Technology)
88 Wood Street
Liverpool, L1 4DQ

Please note, I am only in office until 3pm on a Friday.
tw: @lexy_taylor
t: + 44 (0)151 707 4419
f: + 44 (0)151 707 4445

http://www.fact.co.uk
Admin: +44 (0)151 707 4444
Bookings: +44 (0)871 902 5737

*FACT - Bringing people, art and *
*technology together since 2003*

--------------------------------------------------------------
--------------------------------------------------------------

*Group Therapy: Mental Distress in a Digital Age5 March – 17 May*

Originating from FACT’s extensive work within

mental health and wellbeing, the exhibition

explores the complex relationship between

technology, society, and mental health.

--------------------------------------------------------------
--------------------------------------------------------------

bloors, 2015-04-27:
Hi Ross 
Just to clarify: the window doesn't overlook your space, it's in Will Shannon & Assemble's area (in red). And yes, all sides of the story table can be accessible.
The screen bit is in keeping with the design through the show and is for the info for your project. It'll include the A2 interpretation panel and guide leaflet rack, a TV/monitor with your film on and the map inphographic (on the double boards), so needs to be that scale and placed there. We've deliberately split the two spaces so exhibition visitors can still look at the story table while the workshop sessions are going on in the other area. 
Best,
Tom

www.simonandtombloor.com

cheapjack, 2015-04-28:
Ok I see Tom makes sense in that case let's keep the layout as it is then I think it will work great! In that case lets **go for layout Option 3**

amcewen, 2015-04-28:
I'm happy with @cheapjack's choice of option 3, just wanted to pick up on two points:

First, the pieces which need to be in vitrines are all listed in the "5 steps to a new hand" sections of https://github.com/cheapjack/buildyourown/blob/master/Exhibition/FACT/FACTSpaceSetup.md

Secondly, we're having a TV/monitor to show a film?  What film is that?  Am I forgetting something?  (Quite possible :-)

cheapjack, 2015-04-28:
Yes I did not realise we were showing a film....assumed that might not apply to this as would need to involve so many partners etc.. Seems like could be a fair bit of work that! Is this something that FACT are making like the film made for Cloudmaker?

bloors, 2015-04-28:
Hi
The film is one (we've been told) is being made, one for each exhibitor/group, think it's an interview. Alma knows more about it I think (I hope someone does..!)
Will check over your info, thanks.
Tom 

www.simonandtombloor.com

cheapjack, 2015-04-28:
We all happy with the printers on a freestanding table by as in Option 3? Im also guessing this area of the layout should be pretty much be all pegboard for flexibility hanging storage boxes. Its looking like we will likely have 3 printers now (2 bought, one borrowed) but with each a footprint of  W 338 x D 358 x H 289 mm I think they will be fine.

[image]

bloors, 2015-04-28:
Hi
Actually that’s an old design, sorry for the confusion. The printers are on a wall mounted shelf as originally discussed, see below. 
The shelf is 700mm deep and 2440mm wide so plenty of room for three printers. If you like it could be deeper so there’s more space in front of each printer.

best,
Tom

> We all happy with the printers on a freestanding table by as in Option 3? Im also guessing this area of the layout should be pretty much be all pegboard for flexibility hanging storage boxes. Its [...]

Simon & Tom Bloor
44(0)7866274975
Grand Union Studios
Unit 15 Minerva Works
158 Fazeley Street
Birmingham
B5 5RS
simonandtombloor.com
kiosk1and2.org

bloors, 2015-04-28:
This is the pic, not sure it was sent here
[image]

cheapjack, 2015-04-28:
Great! that works better and pegboard looks good I *think* that's enough pegboard how many hookable boxes did you get in the end?  if they all fit the pegboard then should be fine. Also we should run power up or down the inside of the panels depending on Mark at FACT's recommendations

cheapjack, 2015-04-28:
I've updated the [space setup md file](https://github.com/cheapjack/buildyourown/blob/master/Exhibition/FACT/FACTSpaceSetup.md) now with the correct images

LesleyTaker, 2015-04-29:
It is the film which @almataki mentioned, documenting one element of the
process / a bit of the workspace etc. All of the artists are having some
form of documentation of at least one element of their work. She is trying
to tie down dates with the filmmaker for Liverpool atm.

-- 

Lesley Taker
Curatorial Co-ordinator
FACT (Foundation for Art and Creative Technology)
88 Wood Street
Liverpool, L1 4DQ

Please note, I am only in office until 3pm on a Friday.
tw: @lexy_taylor
t: + 44 (0)151 707 4419
f: + 44 (0)151 707 4445

http://www.fact.co.uk
Admin: +44 (0)151 707 4444
Bookings: +44 (0)871 902 5737

*FACT - Bringing people, art and *
*technology together since 2003*

--------------------------------------------------------------
--------------------------------------------------------------

*Group Therapy: Mental Distress in a Digital Age5 March – 17 May*

Originating from FACT’s extensive work within

mental health and wellbeing, the exhibition

explores the complex relationship between

technology, society, and mental health.

--------------------------------------------------------------
--------------------------------------------------------------

amcewen, 2015-04-29:
Cool, I'm sure the film's been mentioned, but probably didn't lodge in my brain as it wasn't something we needed to deliver :-)

@cheapjack Are we putting all three Ultimakers into the gallery?  I thought the one we were borrowing was mostly to cover the 3-4 months when the other printers are in Norwich so we can finish off any support for the NW participants (and give us some more capacity for printing out hands at the start)?  In which case, there'd only be two going down to Norwich, so do we want to make room for them in the gallery in Liverpool too?

almataki, 2015-04-30:
Hi again. We are working with Jared Schiller http://jaredschiller.co.uk/ and he just confirmed he can come up to Liverpool on May 15th. Might you be around to do some filming with him? talk about the project, DoES, the community and your process?

amcewen, 2015-04-30:
I don't think any of us are around on the 15th, I'm afraid. @cheapjack 
and I are over in Sheffield IIRC, and Patrick generally has childcare 
duties on a Friday.  Is he likely to be up any other dates?

almataki, 2015-04-30:
hmmm would the 16th be any better for you? are you available any other day of w/c 11th May?

cheapjack, 2015-04-30:
My mistake we are in Sheffield on Saturday only: the 16th may, had looked at wrong month somehow so 15th would be ok for me at least as well be in does that day anyway preparing for the Saturday if that would work

almataki, 2015-04-30:
OK this is great news! I wasn't sure Jared would be able to come on a different day and also Will Shannon will be doing concrete casting on Cairns Street that day so it fits very well. Any time that works for you? and is there a number that Jared can use to contact you on the day?

almataki, 2015-05-12:
@cheapjack Jared will be in Liverpool this Friday. Shall I tell him to come and meet you straight at , Gostins Building, 4th Floor. 32-36 Hanover Street. Liverpool L1 4LN ? His number is 07916 168 141 if you need him on the day. What times you'll be around on the day?

cheapjack, 2015-05-12:
Hi mid morning or early afternoon would be ok for me

almataki, 2015-05-13:
Jared arrives at the station 12:20pm so he will be at your studio at 1pm. Is that OK?

cheapjack, 2015-05-13:
That sounds OK by me 1pm is ok maybe could combine with lunch but not sure on availability of @amcewen & @defproc I know @defproc maybe on child care duty that day . it could just be me going over design which is fine by me :) and can pickup any issues my number is 07977406552 for him

amcewen, 2015-05-13:
I should be around then.

almataki, 2015-05-14:
Very small change of plans Jared will be with you about half an hour later as he has to go elsewhere first. hope that's ok.

LesleyTaker, 2015-05-18:
Hi all, just to ask a space related Q: do all of the story plinths need vitrines? Even the iPad? Is this not a problem for if it crashes or anything and it having to be moved? Also, what height were you thinking? Just to reiterate the footprint of the subplinths on the table are 500mm x 420mm FYI

LesleyTaker, 2015-05-20:
Hey, just wondered if I could push you guys a bit on this? I need to order the vitrines by tomorrow in order to make sure we receive them in good time!! Just need confirmation of number and height! L

amcewen, 2015-05-20:
It's four vitrines in total - they're listed in the steps that need them on https://github.com/cheapjack/buildyourown/blob/master/Exhibition/FACT/FACTSpaceSetup.md :-)

So the iPad doesn't need one (we're assuming it'll be tethered to the plinth but something people can pick up and play with), and if they're at least 10cm high that'll be fine (I don't know what sort of sizes vitrines come in...)

closed 2015-05-20 by amcewen

cheapjack, 2015-05-20:
I think Lesley you have what you need now; I've started a [template for vinyl artwork required for the vitrines and plinths in pdf and svg format](https://github.com/cheapjack/buildyourown/tree/master/Exhibition/FACT/interpretation) so @LesleyTaker could you just have a final look at the dimensions of the overall and individual sub-plinths [here](https://github.com/cheapjack/buildyourown/blob/master/Exhibition/FACT/interpretation/Vitrines.pdf) and check they are right so I can layout the vinyl to fit amongst the objects/hand parts etc.
 
We will only need a maximum or 4 vitrines @LesleyTaker and I will be prepping & applying the vinyl to the plinths when they ready. I don't know whether plinths sit over/around the subplinth footprint or not but as long as they give us the minimum height clearance of 10cm (between subplinth and vitrine surface) we will be fine.

I think on a standard entry in to G1 you would approach the vitrines in a way that would mean you'd read Vitrine 1 first: we can always turn the table round if it does not. 

Artwork will live here once Ive finished it
https://github.com/cheapjack/buildyourown/tree/master/Exhibition/FACT/interpretation