cheapjack/buildyourown #13: Make a List of possible people to measure and build

cheapjack, 2015-04-13:


LesleyTaker, 2015-04-15:
Hey, if you want me / Amy to help with this, just CC us into an email with Reach / E-nable? We can chase to get events added to their calendar etc and make sure we have a waiting list of participants?

amcewen, 2015-04-19:
How does the booking system work for other workshops at FACT?  Is it possible to add a comments field to let people indicate that they're bringing someone to get fitted for a hand?  Or would it work better to ask Reach to keep that list and let us know separately?

I spoke to the Reach NW coordinator on Friday, and she's happy to send details out to all their members.

LesleyTaker, 2015-04-20:
@amyjones will make an eventbrite with Sinead from marketing and they will
be able to put everything together in a little e-flyer if that would be ok?
That way the guys can send it out to their networks as is, with whatever
copy they want in the email?

how should we define whether you're a helper or being fitted for / measured
up for a hand? Two separate eventbrites for each event?

-- 

Lesley Taker
Curatorial Co-ordinator
FACT (Foundation for Art and Creative Technology)
88 Wood Street
Liverpool, L1 4DQ

Please note, I am only in office until 3pm on a Friday.
tw: @lexy_taylor
t: + 44 (0)151 707 4419
f: + 44 (0)151 707 4445

http://www.fact.co.uk
Admin: +44 (0)151 707 4444
Bookings: +44 (0)871 902 5737

*FACT - Bringing people, art and *
*technology together since 2003*

--------------------------------------------------------------
--------------------------------------------------------------

*Group Therapy: Mental Distress in a Digital Age5 March – 17 May*

Originating from FACT’s extensive work within

mental health and wellbeing, the exhibition

explores the complex relationship between

technology, society, and mental health.

--------------------------------------------------------------
--------------------------------------------------------------

amcewen, 2015-04-20:
An e-flyer would be cool.  On eventbrite you can have different classes 
of tickets when you sign up can't you?  That'd probably be the best way 
to run it, I think?

LesleyTaker, 2015-04-20:
Sounds good to me. Should we say 14 participants max, 7 of each? or is that
too many?

-- 

Lesley Taker
Curatorial Co-ordinator
FACT (Foundation for Art and Creative Technology)
88 Wood Street
Liverpool, L1 4DQ

Please note, I am only in office until 3pm on a Friday.
tw: @lexy_taylor
t: + 44 (0)151 707 4419
f: + 44 (0)151 707 4445

http://www.fact.co.uk
Admin: +44 (0)151 707 4444
Bookings: +44 (0)871 902 5737

*FACT - Bringing people, art and *
*technology together since 2003*

--------------------------------------------------------------
--------------------------------------------------------------

*Group Therapy: Mental Distress in a Digital Age5 March – 17 May*

Originating from FACT’s extensive work within

mental health and wellbeing, the exhibition

explores the complex relationship between

technology, society, and mental health.

--------------------------------------------------------------
--------------------------------------------------------------

amcewen, 2015-05-20:
This is all in hand now.

closed 2015-05-20 by amcewen