cheapjack/buildyourown #16: How are parts from completed prints stored?

amcewen, 2015-04-13:
During the exhibition the 3d printers will be printing out parts for members of the Reach community.  Each set of prints will be specific to that person, so we need to ensure they're kept together and not confused with others.

The completed parts will also form part of the exhibit, as it shows the process and wealth of options (colours, people involved, etc.)

Obviously they also need to be kept out of reach so that parts don't go missing.  We figured some shelves or similar above the 3d printers would be the best option.  Maybe something like http://www.screwfix.com/p/medium-storage-kit-36-bin/51410 or some shelves and clear-plastic containers so people can see the contents?

bloors, 2015-04-14:
Yes, good–we were also thinking of those wall mounted plastic bins! We'll try to find somewhere that can supply a clear version that can be pegboard mounted. How many might be required?

cheapjack, 2015-04-15:
Ok I think roughly
Tools and useful things : 5
Hands (assuming we had a target of 10) Hands 10
So around 15 I'd say but with flexibilty on sensible order numbers depending on your supplier ;)

closed 2015-04-15 by cheapjack