cheapjack/buildyourown #17: How is spare filament for the 3d printers stored?

amcewen, 2015-04-13:
Similarly to issue #16, there'll be a range of spools of filament for the 3d printers that we'll need to store somewhere, and having it on display will show the range of colours available.

They could easily live on shelves alongside the completed parts, if that's what we go for - or if the parts are stored in bins then we'll need a different solution for these.

bloors, 2015-04-15:
How many spools are there likely to be? We've found a company that can do bins for the hands in 'clear' polypropylene that can be applied to pegboard, they have a good selection of sizes–It'd be good to know the most appropriate for the hands– http://www.shopfittings4u.co.uk/products.asp?catid=116. They also do other bins that might be a good option for the filament etc if the above aren't the right size http://www.shopfittings4u.co.uk/products.asp?catid=113 Again, if you have an idea of the kind of dimensions of bin that might work, let us know so we can check their stock and costs.

amcewen, 2015-04-15:
Good question.  I think we'll be able to have some flexibility on that as we can always store spare spools elsewhere if need be, but ideally we'd have something like 8-12 on display to show the colour options.

I've just measured up one of our existing spools, and they're ~22cm diameter by ~8cm wide.  So the Storbox Big is probably best for the filament.

For the hands, there'll be a palm, cuff, and up to 5 fingers with three pieces each.  The palm and cuff are the big items, roughly 14x14x5cm, so probably the Storbox Cube for each of the hands.  If we can include a few Storbox Bigs for hands too (2-4, depending on aesthetics :-), that'll cover if we get anyone with much bigger hands along.

closed 2015-04-15 by amcewen