cheapjack/buildyourown #18: Do we need to secure the calipers on display?

amcewen, 2015-04-13:
For the first step in "5 steps to a new hand" we'll have a set of measurement calipers and an Android tablet.  

As the tablet is something for people to pick up and interact with, it doesn't feel like a vitrine is right for that step's display.  However, does that mean the calipers are likely to go missing?  Do we need to do anything to prevent that?

cheapjack, 2015-04-13:
Maybe calipers and other things can be secured on the pegboard backing by the printers or above the measurement shelf until needed and gallery assistants help get tools when needed.. Means the gallery assistants (GA's) are a bit like store keepers

amcewen, 2015-04-13:
These aren't calipers for using, they're calipers to be on display to explain the steps you go through to get your new hand.  So that's another reason to secure them on the display, so they don't get stolen to be used in measuring things up for a workshop! :-)

bloors, 2015-04-14:
They should probably be secured, either on pegboard or to the shelf....Linda Brothwell is using the pegboard to display the tools she's made–they will be secured so people can't take them. For your space we could have a mix of secured tools (above '5 steps' shelves) and tools that can be used (above printers or elsewhere). Or we could look at attaching tools to the shelves in some way, but use smaller bits of the coloured MDF to display them in a more museological way...
[image]

DefProc, 2015-04-14:
How's 'wired to the pegboard'?

amcewen, 2015-04-15:
There isn't any pegboard behind the 5 plinths, as that's where the explanation text will be.  Plus I think the calipers should live next to the tablet.

Maybe we can drill a couple of holes in the plinth and then wire/cable-tie the calipers onto the plinth?

DefProc, 2015-06-05:
No calipers on display, just a vinyl.

closed 2015-06-05 by DefProc