cheapjack/buildyourown #9: We need to set a date when the press can get hold of us and who that is

cheapjack, 2015-04-13:
Maybe we can grab a minute this week to set a date for when we happy to talk to press: ie when we get some kind of rough plan #3

amcewen, 2015-04-13:
I wonder if we should set up a desktopprosthetics@doesliverpool.com and have that go to a special group in Support Bee where the three of us will get emails from it?  That will hopefully make it easier to manage things (and we can book meetings/calls/etc. into the calendar?)

cheapjack, 2015-04-13:
Yes please would you mind setting that up; realised I could have set this up through DoES on github actually

amcewen, 2015-04-13:
Bah, we'd need to upgrade our SupportBee plan for that, which would cost an extra $40/month.  Not sure it's worth it for that.

Anyway, I've created an email alias desktopprosthetics@doesliverpool.com which will forward to the three of us, so at least it's easier to let people know how to contact us all.

cheapjack, 2015-04-13:
poo. Cheers makes sense!

R

closed 2015-05-15 by amcewen