cheapjack/buildyourown #20: How do members of the public try out wearing a prosthetic hand?

amcewen, 2015-04-15:
We'd mentioned having some stuff for people to try picking up, etc. 

What are those things?  Where do they live?  On the table in the middle?

cheapjack, 2015-04-15:
I think on the table, but with a storage box set apart from the others & labelled, so
* Printed 'example' hand
* finishing/clipping and sanding tools
* Custom 3d printed dowels for finishing: we had initially talked about an UP printer (which prints with deifferent spec material) on the table making these useful tools

LesleyTaker, 2015-04-16:
Do we still want the UP? Love the idea of having useful bits on the table.
Hannah, the curator from Norwich, had worries about them going missing
though, so maybe we have a way to attach? Or any ideas around this?

-- 

Lesley Taker
Curatorial Co-ordinator
FACT (Foundation for Art and Creative Technology)
88 Wood Street
Liverpool, L1 4DQ

Please note, I am only in office until 3pm on a Friday.
tw: @lexy_taylor
t: + 44 (0)151 707 4419
f: + 44 (0)151 707 4445

http://www.fact.co.uk
Admin: +44 (0)151 707 4444
Bookings: +44 (0)871 902 5737

*FACT - Bringing people, art and *
*technology together since 2003*

--------------------------------------------------------------
--------------------------------------------------------------

*Group Therapy: Mental Distress in a Digital Age5 March – 17 May*

Originating from FACT’s extensive work within

mental health and wellbeing, the exhibition

explores the complex relationship between

technology, society, and mental health.

--------------------------------------------------------------
--------------------------------------------------------------

cheapjack, 2015-04-28:
I think we do... even though now we should have a borrowed 3rd ultimaker printer we need a list of the tools but maybe wont know what these will be until we start printing the example hand as soon as we get the printers, unless Jason Abbott had any suggestions @amcewen meeting him tomorrow so if he sees this then we could ask. 

Sanding blocks and dowels you can wrap sandpaper in seem useful need to make a few designs but could include a hoop on each that stays clipped to the table for security maybe...for small items if they go missing, then we print more and it gives another thing printing for people to watch...or when they not used the go in the pegboard hanging box for tools. Actually not many people even think you need tools to finish off some printed stuff; so be good to show that.

LesleyTaker, 2015-04-29:
We can always still use the UP but locate it in the FACT Lab in G2? Just
for safety / space sake?

-- 

Lesley Taker
Curatorial Co-ordinator
FACT (Foundation for Art and Creative Technology)
88 Wood Street
Liverpool, L1 4DQ

Please note, I am only in office until 3pm on a Friday.
tw: @lexy_taylor
t: + 44 (0)151 707 4419
f: + 44 (0)151 707 4445

http://www.fact.co.uk
Admin: +44 (0)151 707 4444
Bookings: +44 (0)871 902 5737

*FACT - Bringing people, art and *
*technology together since 2003*

--------------------------------------------------------------
--------------------------------------------------------------

*Group Therapy: Mental Distress in a Digital Age5 March – 17 May*

Originating from FACT’s extensive work within

mental health and wellbeing, the exhibition

explores the complex relationship between

technology, society, and mental health.

--------------------------------------------------------------
--------------------------------------------------------------

amcewen, 2015-05-30:
Okay, we're printing off some hands in different sizes, so that people can try them on.

We'll have at least one at 65mm (Baylee-sized), one at 90mm (demonstration_adult), one at 80mm.

cheapjack, 2015-05-31:
Drem Murray will be lending us some other designs also so we will have plenty of things for the public to see and try out, managed by the gallery guides of course.

closed 2015-05-31 by cheapjack