cheapjack/buildyourown #21: Exhibition Text Panel

almataki, 2015-04-15:
I'd just like to flag sth up. 
As you can imagine the curators are putting together texts/captions for the projects in the show and for the gallery guide etc. These are a bit different to the marketing print and aim to provide a wider context/background to the work (as all exhibition texts do). We are still editing those heavily but I plan to share them with you on Monday next week (20th April) advance of sign off  for Friday (24th April). 
Each project gets one panel of about 150words + credits and urls and then the same text will be used in the printed gallery guide together with biogs. All this to be shared with you on Monday. The text will be based on your copy from README and does not reference Baylee.

almataki, 2015-04-15:
To add to this the panel texts will be etched into 3mm engraving laminate-layers of white and black acrylic stuck together. An engraver then cuts down through the black to leave a relief of white text. The panels will be attached to a pegboard panel. So each project will have at least a single width pegboard with a white letters in black acrylic panel about A2 in size.

amcewen, 2015-04-15:
Cool.  That sounds good.  Is that laser-engraved or CNC milled, do you know?  (Just curious :-)

almataki, 2015-04-17:
Adrian I asked the guys to clarify but haven't heard back. 
In the meantime, can we start with your biog? I drafted the below but it should be a bit longer and it would be nice to include some of your past projects. We've done this for the other artists' biogs listing a few older exhibition etc. However it should be longer than 80-100words tops.

DoES Liverpool
DoES Liverpool is a community of makers and entrepreneurs with a diverse range of skills of interests. DoES founding members include creative technologists and developers Adrian McEwen, John McKerrell, Hakim Cassimally, Paul Freeman, Andy Goodwin and Ross Jones. The community includes a co-working space, workshop, event hub, and much more. *[to add past projects]* www.doesliverpool.com

Do you think we should include a biog- or more like an about text for e-NABLE?

Thanks!

almataki, 2015-04-17:
correction: It should NOT be longer than 80-100 words tops

almataki, 2015-04-17:
OK so this is where the caption text currently stands, please check the crediting and let me know if you'd like to add anything. The caption shouldn't be more than 200words.

**Desktop Prosthetics (2015)**
DoESLiverpool: Ross Dalziel, Patrick Fenner and Adrian McEwen
??E-nabling the Future network??
??Norwich Hackspace??

Hacker and entrepreneur community DoESLiverpool presents an iteration of the open-source project E-nabling the Future, a project that allows people across the world to build prosthetic hands using 3D printers. e-Nable is a worldwide movement of tinkerers, engineers, 3D print enthusiasts, occupational therapists, university professors, designers, artists and families who work together to build simple, low-cost devices for people in need of prosthetic limbs or to guide them as they build one themselves. 
Inspired by a local family who used DoESLiverpool's workshop to print hand parts independently, DoES has installed a ‘production line’ for prosthetic hands using 3D printing technology and open-source designs. *[more about the ethos of DoES, the philosophy of Open Source, Creative Comms and decentralising production and bringing communities together to teach one another and trade skills in an equal, free-trading style environment.]*

Over the course of the exhibition, DoES in Liverpool and Norwich Hackspace invite you to get involved in testing and building prosthetics, experiment and understand what prosthetics are and how access to rapid prototyping and open source digital making tools can help people make the world the way they want it. Check the events panel for more details of how you can take part.
*[also add something for visualisation? – CC needs to confirm this]*

*[other credits]- This is for anyone or any other organisation you’d like to credit and it’s not mentioned in the text*

www.doesliverpool.com
www.enablingthefuture.org

almataki, 2015-04-17:
And sorry this is all done in a hurry and I'm well aware you guys have a lot on your plate already, but ideally we want to have the text proofed next week so if you could please send me your comments/edits etc by Tuesday that would awesome and very much appreciated! Thanks!

cheapjack, 2015-04-17:
Ok Ive put all this text [here](https://github.com/cheapjack/buildyourown/blob/master/Exhibition/FACT/interpretation/MainTextPanelText.md) and started to edit it for next week. By then we will have run over things with Reach, pinged Enabling the future but would be good to have got in touch with Ultimaker so we could include their logo? @DefProc can we get in touch with them about this (think you have contact there) before we see them at Maker Faire next week?
 
Ive included tick boxes so that we can see whether the partners inclusion is ok to go. Logo folder needs populating

cheapjack, 2015-04-20:
Can we have a look at this tomorrow, im working elsewhere till evening its the draft of the exhibition panel, will try have another crack at it if I get some time in the morning; if we are directly referring to Norwich hackspace do we need their profile? I think not as this can take place at a modified panel their end? Comments welcome!

So [draft is here](https://github.com/cheapjack/buildyourown/blob/master/Exhibition/FACT/interpretation/MainTextPanelText.md)

amcewen, 2015-04-20:
I've added the DoES logo.  The text is looking good so far.  I wonder if we should include something to encourage people to donate to Reach if they want to help out financially?

almataki, 2015-04-21:
I think the text is looking good but it won't be possible to present it exactly in this format. All panels follow the same model stylistically:
Title (year), Artist, main text (2-3paras), additional credits in italics and relevant urls
We will not include the biogs in the panels but in the gallery guide - staple stitched booklet - which is going to be available for free as you enter the gallery. The guide will include the same text plus any relevant biogs. Would you mind if I edit the draft to reflect the layout of the exhibition graphics?

almataki, 2015-04-21:
I think it will look neater to include logos on a separate panel as in 'here is everyone behind this project'

almataki, 2015-04-21:
Hi again, so I tweaked the text a bit and structured it so it can be laid out like the other captions. What do you think? 

**Desktop Prosthetics (2015)**
DoESLiverpool: Ross Dalziel, Patrick Fenner and Adrian McEwen

DoESLiverpool presents an iteration of the open-source project Enabling the Future, a global network of passionate volunteers using 3D printing to give the world a 'helping hand'. Inspired by a local family who used DoESLiverpool's workshop to print hand parts independently, DoES has installed a ‘production line’ for prosthetic hands. They will be using 3D printing technology and the open-source "Raptor Hand" design, which was developed to make printing and assembly of the hand’s components easier for their makers and recipients.

*Desktop Prosthetics* aims to raise awareness of some of the more practical uses of 3D printing and demonstrates how collaborative communities like DoES and Enabling the Future work together - sharing skills and specialised knowledge to solve real problems.
*[Also add something for visualisation? – CC needs to confirm this]*

Over the course of the exhibition, DoES Liverpool will be working with Reach, the association for children with upper limb deficiency, to help a number of local children build and assemble a prosthetic device. DoES invite you to get involved in testing and building prosthetics, experiment and understand what prosthetics are and how access to rapid prototyping and open source digital making tools can help people make the world the way they want it. Check the events panel for more details on how to take part. 

*Developed with the support of Reach North West and E-nabling the Future* 

www.doesliverpool.com 
www.enablingthefuture.org
www.reach.org.uk 

[for gallery guide same as the above + biogs below]
**DoES Liverpool**
DoES Liverpool is a community of makers and entrepreneurs with a diverse range of skills and interests. It was founded by creative technologists and developers Adrian McEwen, John McKerrell, Hakim Cassimally, Paul Freeman, Andy Goodwin and Ross Jones with the intention to create a friendly environment to foster collaboration and support the growth of the region in a meaningful way – by doing. The community includes a co-working space, a workshop and an event hub. Any profit helps and sustains the community. www.doesliverpool.com  

**e-NABLE**
e-NABLE first began in 2013, consisting of about 300 people who owned 3D printers or who had design skills to share, and who wanted to help improve the first open sourced design for 3D printed hands that had been released online. They also wanted to help print and assemble free 3D printed prosthetic devices to anyone that reached out to them. Now, in 2015, their “matching system” has over 1000 recipients in various stages and 3000 registered volunteers, who all work to make more people’s lives easier thanks to 3D printed hands. www.enablingthefuture.org

cheapjack, 2015-04-21:
This looks good but I prefer the bullet points of what is DoESLiverpool though and not sure if we need to list the founders but I think @amcewen can speak more for DoES than me; what do you think? I wonder if we'd need to get someone from Enable to agree to this though and I dont think we have a direct contact yet

almataki, 2015-04-22:
@cheapjack and @amcewen I think bullet points will look odd when the rest is all text. Here is a page of how the gallery guide's pages are going to look like.
[image]

cheapjack, 2015-04-22:
Ok fair enough ive rewritten and included your text for @amcewen and @DefProc to look over
So have a [look here](https://github.com/cheapjack/buildyourown/blob/master/Exhibition/FACT/interpretation/MainTextPanelText.md)

almataki, 2015-04-22:
Thanks! I will forward it to my proof reader once @amcewen and @DefProc also look over. Do you think it's better not to include a biog for enable? since we have no contact with them. All text has been taken from their website so i can't see why they'd mind.

almataki, 2015-04-27:
In relation to the guide AES also suggested there is a page with one photo per project - see mock up below. Can we use Baylee's  photo for this? Of course properly credited and everything.
[image]

almataki, 2015-04-27:
Also the curators were keen that your voice is somehow captured in the exhibition. So we would like to print a quote by all artists – about 30words for each commission. Ideally something that brings alive your process. If answering a direct question helps, we thought maybe answering this ‘How collaboration and skill-sharing influenced your process in developing the work?’ or ‘Describe in a few words the processes that went into making the work?’ could be good starting points?
Might you be able to send me a combined quote/reply to one the questions this week?

Trying to reduce your workload, I was thinking that maybe we can use one of your answers to the questions by OPEN. For example,
"DoESLiverpool like many makerspaces and hackspaces, as an organisation we do not do projects in a sense, we simply help our community to realise their projects with the facilities we have. Our community is essentially anyone who wants to do something with us. "

What do your think?

cheapjack, 2015-04-27:
RE open and workload; exactly we are using those answers as a template FAQ which you can [see here](https://github.com/cheapjack/buildyourown/blob/master/FAQ.md)

cheapjack, 2015-04-28:
Hi just checking we are all happy with the exhibition panel text so far? Its [here](https://github.com/cheapjack/buildyourown/blob/master/Exhibition/FACT/interpretation/MainTextPanelText.md)

Also regards the voice of us I've added some more detailed stuff about maker spaces and responded @almataki question  "to describe in a few words the processes that went into making the work" at the bottom of the FAQ; ive just waded in on that one, please comment and edit if you want @amcewen @DefProc [have a look here](https://github.com/cheapjack/buildyourown/blob/master/FAQ.md)

amcewen, 2015-04-29:
The text looks great to me.  I made a minor tweak to remove the "North West" from "Reach North West" because (a) the NW bit is mostly a regional coordinator rather than a full office, and (b) we're also talking to Reach head office and will be aiming to hook up Norwich Hackspace with the local Reach coordinator there.

amcewen, 2015-04-29:
The FAQ looks good (and we'll be continually updating it with new questions anyway).  I updated the link to Nesta's report as it's been published now.

almataki, 2015-04-30:
Hi guys and thanks for looking through the texts. I had the text proofed and my proof reader sent the following comment:
"I've just been having another look at the e-NABLE website and it looks to me as though e-NABLE is the official organisation name and Enabling the Future is more a kind of tag line. So I think perhaps it would be better to go with e-NABLE throughout. What do you think?"
I have to say the two names have been confusing when we talk about the project and I agree with her that we should choose one way of referring to e-nable throughout. do you see any issue with that?

cheapjack, 2015-04-30:
Well spotted and I'd agree with that. @DefProc has had some contact with them so will follow that up with him and get in touch with them asap. Im sure it will be fine to take from their pack but we should get some direct contact and thats admittedly well overdue

almataki, 2015-05-01:
Great! text and quote are signed off so have been sent to the proof reader to ensure coherence throughout the exhibition.
The last thing for the interpretation is the photo for the guide. Is it OK for us to use Baylee's photo?
Thanks!

amcewen, 2015-05-02:
Assuming you mean this photo - https://github.com/cheapjack/buildyourown/blob/master/RaptorHandBeingWorn.jpg - then yes, that's fine.  Just needs to be credited to Jason Abbott.

almataki, 2015-05-05:
Yes that's right! will make sure it's credited!

almataki, 2015-05-05:
I just saw your comment about the raptor hand. I added the line to the exhibition text below. Today and tomorrow is the very last opportunity for any changes to the exhibition text. In practice we can only add rather than edit as the text has already been proofed.

**Desktop Prosthetics (2015)**
DoESLiverpool: Ross Dalziel, Patrick Fenner and Adrian McEwen
 
DoESLiverpool presents an iteration of the open-source project e-NABLE, a global network of passionate volunteers using 3D printing to make prosthetic hands. Inspired by a local family who used DoESLiverpool's workshop to print hand parts, DoES has installed a ‘production line’ for prosthetic hands. They will be 3D printing the open-source Raptor Hand design, which is more of a tool rather than a fully functional prosthetic device, developed to make printing and assembly of the hand’s components as simple as possible.
 
Desktop Prosthetics aims to raise awareness of some of the practical uses of 3D printing and to demonstrate how collaborative communities like DoES and e-NABLE work, sharing skills and specialised knowledge to solve problems.
 
Over the course of the exhibition, DoESLiverpool will be working with Reach, the association for children with upper limb deficiency, to help a number of local children build and assemble a prosthetic device. DoES invites you to get involved with testing and building prosthetics, to better understand what prosthetics are, and to see how access to rapid prototyping and open-source digital-making tools can help people make the world the way they want it. Check the events panel for more details on how to take part.
 
*Developed with the support of Reach and e-NABLE.*
 
www.doesliverpool.com
www.enablingthefuture.org
www.reach.org.uk

amcewen, 2015-05-05:
Looks good to me.

closed 2015-05-05 by almataki