cheapjack/buildyourown #22: Additional Interpretation

almataki, 2015-04-15:
I understand that additionally to the gallery interpretation #21  you are keen to have explanation texts around the printers, prosthetic components etc. It makes sense that there is more explanatory text around your display as it is quite different to everyone else's but as you can also imagine we have to ensure that the exhibition has a coherent look to it. So a couple questions in relation to this.
Where you planning to design these graphics yourselves?
How were you thinking of presenting them?

amcewen, 2015-04-19:
I don't think we've got as far as working out exactly how we're going to do that yet :-)

However, we should be able to produce something similar to the panels you're doing for the rest of the show.  Will you be able to share the source files, and/or details of things like the fonts used?  Then we can use them as a basis for ours.

It'll probably be 1.5mm acrylic rather than 3mm, but we can laser-etch that in-house - https://www.flickr.com/photos/amcewen/16215574363/ shows an example I made that way recently.  Obviously that's black text on white, whereas for the exhibition it would be the other way round, but that's just a case of buying the right laminate acrylic :-)

almataki, 2015-04-21:
I don't think this element should use the same design as the exhibition design, I'd be keen that it is distinct from the overall design and provided a cohesive visual identity to your project/area. 
However it is also important it doesn't clash with everything else so maybe we can chat about it once you know what type of info you want to include? Wasn't there an idea originally to use vinyls for this? It would also be nice to include An Endless Supply in this discussion.

almataki, 2015-04-22:
I spoke to Robin from An Endless Supply for the additional interpretation and they're happy to work on this so they have an overview of the overall exhibition aesthetic- of course all in discussion with you. When do you think you'll know the extend of the copy for this?

almataki, 2015-05-14:
Hi guys please see below the spread for Desktop Prosthetics from the latest gallery guide design. The exploded view looks good!
[image]

cheapjack, 2015-05-21:
@almataki looks great any chance you can get me the original How to artwork and font so we can match it on the sub-plinth? Ideally in some kind of editable vector format?

LesleyTaker, 2015-05-21:
This might also be a good time to talk about the Ultimaker logo etc which Daniel mentioned in his email to @DefProc and you guys? Could do it at the same time?

cheapjack, 2015-05-22:
Hi it may make sense to use the pegboard area around the printer station to display their logo: then we don't need an exact match to the formatting of the main interpretation and then it's clear what they are supporting.

Also need the BYO font still if anyone has it :)

almataki, 2015-05-22:
will get that to you Ross! sorry for the delay, I've asked Robin to send it over.

almataki, 2015-05-26:
Hi and sorry again for the delay. I had a chat with Robin and we think that rather than trying to match the exhibition design it makes more sense for the instructions to be designed in your own style so it is distinct to the rest and specific to your project. I need however to ask you to please share them with me before you print them on vinyl, just for us to have the general overview of the display. Hope that makes sense. Thanks!

cheapjack, 2015-05-26:
Ok well will just choose a generic font then. But it will mean it's the only different font if you dont mind. Will pick something like arial or one of these http://openfontlibrary.org

almataki, 2015-05-26:
Yes one of these will look good!

LesleyTaker, 2015-05-26:
I love Glacial Indifference. Mostly for the name...

-- 

Lesley Taker
Curatorial Co-ordinator
FACT (Foundation for Art and Creative Technology)
88 Wood Street
Liverpool, L1 4DQ

Please note, I am only in office until 3pm on a Friday.
tw: @lexy_taylor
t: + 44 (0)151 707 4419
f: + 44 (0)151 707 4445

http://www.fact.co.uk
Admin: +44 (0)151 707 4444
Bookings: +44 (0)871 902 5737

*FACT - Bringing people, art and *
*technology together since 2003*

--------------------------------------------------------------
--------------------------------------------------------------

*Build Your Own: Tools for Sharing4 June – 31 August*

Showcasing four new commissions that span different

models of creative practice, including art, craft, and

creative technology, *Build Your Own* celebrates a new

spirit of generosity through making from your local

street to global maker networks.

--------------------------------------------------------------
--------------------------------------------------------------

closed 2015-07-01 by cheapjack