cheapjack/buildyourown #23: Workshop/events programme

amyjones, 2015-04-16:
Hi all, 

Just want to go through what the suggested dates for workshops would be from our side:  

June 5th:   Artist Talks (representative(s) from DoES) 

June 6th:  Measuring workshop - We do have another workshop on this day but I don't see that being an issue and understand it's important to get this process started asap.  

June 13th:   I'd like to recommend this for the prosthetics workshops as there is a conflicting workshop on June 20th.  Can you let me know asap if the 13th?  We may still be able to get it in the brochure before it goes to print if you let me know today! 

Assembly workshop Saturday 18th July & Saturday 15th August
TBD:  These look fine for now.  I'm compiling a Public Programme calendar today so we should get a better sense of what will fit best in July and August - will get back to you to 100% confirm!  

For now can you just confirm the dates of 5th, 6th and 13th June are workable from your side?  

Thanks so much! 

Amy

cheapjack, 2015-04-16:
Amy have a look at [the event page](https://github.com/cheapjack/buildyourown/tree/master/events) I've placed all the dates there and drafting the copy for each one in folders for each kind of event

Sorry not enough of us can manage the 13th: we've now suggested the 21st on the Sunday for the [What's Your Super Power: Prosthetic Hack](https://github.com/cheapjack/buildyourown/blob/master/events/prostheticshack/ProstheticHack.md); would that work?

cheapjack, 2015-04-16:
I've also changed this issue so it sounds more like a task that we can tick off and resolve rather than just an area of activity that needs to be thought about :)

amyjones, 2015-04-16:
Hi Ross,

Ok, I'm working around a jam packed programme here so excuse the back and
forth!! How does the following work:

Measuring workshop: Sat June 6th

Assembly workshops: 1st one - Sorry 18th not available so would you prefer
Sat 11th July or Sat 25th July?
2nd session -  Sat 15th August

Having the Prosthetic Hack day in June is going to be difficult - seems a
shame for it to share the audience with one of our other events and Sunday
is not a popular day for this kind of activity in our experience.  Below is
a selection of dates which would mean the event would get top billing!

Sat July 11th/25th (depending which you choose for the assembly workshop)
Sat 8th August
Sat 22nd August

Could you confirm a date for the first assembly workshop and for the Hack
day asap?

That's great about the copy for the measuring up and hack session, thank
you!  Do you have anything for the assembly workshops? Can't see it on
Github.

 I also need the following info from you:

   - Age suitability of each session (e.g 12+)
   - maximum number of participants per session
   - Detailed breakdown of the activity taking place in each session (inc.
   the duration of each element and breaks if required)
   - Times/duration of each session - our Saturday programme is typically
   12pm - 4.30pm
   - Location of workshops - are they ok to happen by your work in Gallery
   1?

I really need this by Monday at the very latest (ideally tomorrow!) if
we're to include the June activities in the Brochure.   Also, we'll need to
know the relevant logos/credits to be used in the Brochure.  Could you send
these through?

Sorry lots of questions/actionables in there!  Promise my messages will be
much shorter in the future!!

Amy

-- 

---
Amy Jones
Exhibition Coordinator
FACT (Foundation for Art and Creative Technology)
88 Wood Street
Liverpool, L1 4DQ

t: +44 (0)151 707 4403

www.fact.co.uk

Admin: +44 (0)151 707 4444
Bookings: +44 (0)871 902 5737
Information: +44 (0)151 707 4464

*FACT - Bringing people, art and *
*technology together since 2003*

--------------------------------------------------------------
--------------------------------------------------------------

*Group Therapy: Mental Distress in a Digital Age5 March – 17 May*

Originating from FACT’s extensive work within

mental health and wellbeing, the exhibition

explores the complex relationship between

technology, society, and mental health.

--------------------------------------------------------------
--------------------------------------------------------------

amcewen, 2015-04-16:
For future ref. everyone, this would've been better as comments on issue #6 - we can reopen closed issues if things change, and that would just keep all the event date discussion in one place.  (At least by mentioning it now there'll be a link between them :-)

cheapjack, 2015-04-16:
Ok I've considered clashes in all our personal diaries and DoES Events and the dates you suggest amy, so they are now updated [here](https://github.com/cheapjack/buildyourown/blob/master/events/README.md) and are:

 * [Measuring Up](https://github.com/cheapjack/buildyourown/tree/master/events/measuringupworkshop) **Saturday 6th June**
 * [What's your superpower? A Prosthetic Hack!](prostheticshack) **Saturday 25th July** 
 * Assembly workshop **Saturday 8th August** & **Saturday 22nd August**
 * TBD

@amcewen is following up on the other details @amyjones ta

amcewen, 2015-04-16:
Splitting the extra questions out into their own issues.  So the greater detail we'll need for the brochure is in issue #24 and then the stuff to work out nearer to the events running is in issue #25

amcewen, 2015-04-17:
I've spoken to Reach and those dates should be okay for them (no guarantees, but no obvious clashes :-)  Haven't been able to get in touch with Jason, but hopefully he'll be able to make them too.

cheapjack, 2015-04-17:
great

cheapjack, 2015-04-17:
Ive updated the event section dates, added some text and images if required and a tick box to tick when we are happy with it all

amyjones, 2015-04-20:
Perfect, thanks! Those dates are super.

-- 

---
Amy Jones
Exhibition Coordinator
FACT (Foundation for Art and Creative Technology)
88 Wood Street
Liverpool, L1 4DQ

t: +44 (0)151 707 4403

www.fact.co.uk

Admin: +44 (0)151 707 4444
Bookings: +44 (0)871 902 5737
Information: +44 (0)151 707 4464

*FACT - Bringing people, art and *
*technology together since 2003*

--------------------------------------------------------------
--------------------------------------------------------------

*Group Therapy: Mental Distress in a Digital Age5 March – 17 May*

Originating from FACT’s extensive work within

mental health and wellbeing, the exhibition

explores the complex relationship between

technology, society, and mental health.

--------------------------------------------------------------
--------------------------------------------------------------

amcewen, 2015-04-20:
Great.

closed 2015-04-20 by amcewen