cheapjack/buildyourown #24: Provide brochure detail for events

amcewen, 2015-04-16:
For each event we need to know
 * [ ] Age suitability of each session (e.g 12+)
 * [ ] Times/duration of each session - our Saturday programme is typically 12pm - 4.30pm
 * [ ] Location of workshops - are they ok to happen by your work in Gallery 1?

amcewen, 2015-04-19:
Age-wise, chatting to Jason about it, he reckons anyone above 3 would likely get some benefit from the hands.  However, is this more to encourage who should/shouldn't along separately from attendees in the Reach network?

@cheapjack I see you've stuck 11-4 for the "What's your superpower?" workshop, shall we run with that time for all of them?

If we're expecting to have about a dozen people who need hands along for the measuring up workshop, is there going to be enough space around a single table?  Is there scope for putting an extra table in Gallery 1 during the workshop? It would be good to have it next to the printers, etc. so it'd be best if we didn't have to move it upstairs...

cheapjack, 2015-04-19:
I guess we should say 3 and above then, children accompanied by an adult. Yes I think we go 11 - 4 for everything if Amy & Lesley agree

cheapjack, 2015-04-19:
Actually we should follow 12 -4.30 as that will be what they generally do for saturday work

LesleyTaker, 2015-04-20:
12-4.30 sounds great, guys. And all the copy looks fine for the brochure, do you want to look over @amyjones. Will send it over to Marketing. Might be good to resolve how we invite participants as being helpers / receivers of hands, as suggested on the other issue, but that's it really! We can make this clear through the eventbrite and direct people to that from the website.

amyjones, 2015-04-20:
Hi Ross,

Yep those times are perfect, thank you! Can you flesh out the Prosthetic
Hack day text a little more for me? Just because this one isn't quite as
self explanatory as the measuring and assembly ones (clues in the title
with those!)

Having those three bullet points with more detail about the structure of
the workshops (as you've done for the Measuring workshops) is super
useful.  If you could add that in for both Prosthetic Hack that would be
super!

And just to follow up on Lesley's question about the number of participants
- will it be the same number of participants for each workshop?  Would we
have slightly more for the Prosthetic Hack and assembly sessions?

Thanks so much for all your work on this!!

Amy

-- 

---
Amy Jones
Exhibition Coordinator
FACT (Foundation for Art and Creative Technology)
88 Wood Street
Liverpool, L1 4DQ

t: +44 (0)151 707 4403

www.fact.co.uk

Admin: +44 (0)151 707 4444
Bookings: +44 (0)871 902 5737
Information: +44 (0)151 707 4464

*FACT - Bringing people, art and *
*technology together since 2003*

--------------------------------------------------------------
--------------------------------------------------------------

*Group Therapy: Mental Distress in a Digital Age5 March – 17 May*

Originating from FACT’s extensive work within

mental health and wellbeing, the exhibition

explores the complex relationship between

technology, society, and mental health.

--------------------------------------------------------------
--------------------------------------------------------------

cheapjack, 2015-04-20:
Hi I think numbers could more for the Prosthetic hack but I think 10 might be a general maximum; above that it may need to split into G2 though I like it kept around the 3d printer queue and G1 exhibition in general for context but guess it can start there and then shift.

Also Ive [updated the copy](https://github.com/cheapjack/buildyourown/blob/master/events/prostheticshack/ProstheticHack.md) so it explains the content of the day. 
Im assuming the bullet points under Workshop Activity arent part of the copy for the brochure but a guide on content for us all.

Id like to sign this and the other workshop copy off asap now Reach are happy with dates; Adrian can you comment if the descriptions look ok; especially the prosthetic hack. I dont want to be too prescriptive on how the day would run so ive not promised we would print stuff and we may want to prototype in modelling clay and software maybe so that planning detail can come later.

Also would like to close this issue and stop worrying about brochure if everyone happy!

amcewen, 2015-04-20:
The copy looks good to me.

On the workshop numbers, I think we might need a slightly bigger group for the measuring up one, as we'll want about 10-15 people who need hands to be there to fill our "order book" for the printing :-)  And they'll need a parent/guardian to accompany them.

Does eventbrite let you set up a waitlist?  I guess we can see how things go and if there's big demand maybe we can split the measuring workshop into two shifts to get more people through?

amyjones, 2015-04-21:
Copy looks good! Let's close that off!

Re: eventbrite - I'm not much of an expert with it myself but Sinead, my
colleague in marketing knows all its secrets and is back off annual leave
tomorrow so we can discuss that further then.

Is we go for a bigger group of Max 15 - is that still ok to happen in G1?
There's a possibility there will be other activity in G2 (we have a number
artist residencies over the period of the exhibition) so if it's likely
we'll use G2 we'll need to book that out in advance.

-- 

---
Amy Jones
Exhibition Coordinator
FACT (Foundation for Art and Creative Technology)
88 Wood Street
Liverpool, L1 4DQ

t: +44 (0)151 707 4403

www.fact.co.uk

Admin: +44 (0)151 707 4444
Bookings: +44 (0)871 902 5737
Information: +44 (0)151 707 4464

*FACT - Bringing people, art and *
*technology together since 2003*

--------------------------------------------------------------
--------------------------------------------------------------

*Group Therapy: Mental Distress in a Digital Age5 March – 17 May*

Originating from FACT’s extensive work within

mental health and wellbeing, the exhibition

explores the complex relationship between

technology, society, and mental health.

--------------------------------------------------------------
--------------------------------------------------------------

cheapjack, 2015-04-22:
Lets say we need G2 for the Measuring workshop to hold 15 people max, as over 10 maybe difficult to manage in just G1: so we can have enough people for our 'order book' ie hands measured to print. Then I think have G2 for Prosthetic Hack overspill. So the assembly days can just be in G1. So can we make sure those dates we've agreed on that would use G2 also wont clash? ie Saturday 6th June Measuring, then Saturday 25th July for the Prosthetic Hack!

amyjones, 2015-04-22:
I'm afraid we have another workshop in G2 on June 6th but we can have G2
for overspill for Prosthetic Hack.  Can you make just G1 work for the
measuring workshop do you think?

A

-- 

---
Amy Jones
Exhibition Coordinator
FACT (Foundation for Art and Creative Technology)
88 Wood Street
Liverpool, L1 4DQ

t: +44 (0)151 707 4403

www.fact.co.uk

Admin: +44 (0)151 707 4444
Bookings: +44 (0)871 902 5737
Information: +44 (0)151 707 4464

*FACT - Bringing people, art and *
*technology together since 2003*

--------------------------------------------------------------
--------------------------------------------------------------

*Group Therapy: Mental Distress in a Digital Age5 March – 17 May*

Originating from FACT’s extensive work within

mental health and wellbeing, the exhibition

explores the complex relationship between

technology, society, and mental health.

--------------------------------------------------------------
--------------------------------------------------------------

amyjones, 2015-04-24:
Just want to confirm with you the split of those getting measured and those
helping out for the measuring workshop - are we hoping to measure 10 people
during this session?

If so seems like we could easily expect another 10 people (particularly
keeping in mind children and their parents) as helpers. This would bring
the total number of participants to 20 and we do not have G2 as overspill
for this session (However the FACT connect space could be an option if we
really need extra space!)

Let me know your thoughts!

Amy

-- 

---
Amy Jones
Exhibition Coordinator
FACT (Foundation for Art and Creative Technology)
88 Wood Street
Liverpool, L1 4DQ

t: +44 (0)151 707 4403

www.fact.co.uk

Admin: +44 (0)151 707 4444
Bookings: +44 (0)871 902 5737
Information: +44 (0)151 707 4464

*FACT - Bringing people, art and *
*technology together since 2003*

--------------------------------------------------------------
--------------------------------------------------------------

*Group Therapy: Mental Distress in a Digital Age5 March – 17 May*

Originating from FACT’s extensive work within

mental health and wellbeing, the exhibition

explores the complex relationship between

technology, society, and mental health.

--------------------------------------------------------------
--------------------------------------------------------------

cheapjack, 2015-04-30:
On managing the numbers for the measuring workshop: I think with no G2 (which we didnt know about at the time when we agreed on the event dates which we know were tricky to pin down for everyone) we would need to do as much as we could in and around G1 potentially spilling out into the rest of the show in G1 and into the lobby. We may have to use parts of the cafe; could we reserve that area of the cafe closest to the toilets:? Im sure participants would buy stuff from Leaf while they are thre. We talked to Jason Abbott (the father of Baylee) about it and as a group and we thought we could just manage it like that: possibly asking people to wait. Its a fairly quick process to measure anyway: and people can still explore the rest of the work. Do you think this is an ok plan? I cant see where else date wise this can go considering we need to do it at the start of the project

amyjones, 2015-04-30:
Hi Ross,

Yes, no problem to book that space with them. Will do this for you this
afternoon.

Very best,

Amy

-- 

---
Amy Jones
Exhibition Coordinator
FACT (Foundation for Art and Creative Technology)
88 Wood Street
Liverpool, L1 4DQ

t: +44 (0)151 707 4403

www.fact.co.uk

Admin: +44 (0)151 707 4444
Bookings: +44 (0)871 902 5737
Information: +44 (0)151 707 4464

*FACT - Bringing people, art and *
*technology together since 2003*

--------------------------------------------------------------
--------------------------------------------------------------

*Group Therapy: Mental Distress in a Digital Age5 March – 17 May*

Originating from FACT’s extensive work within

mental health and wellbeing, the exhibition

explores the complex relationship between

technology, society, and mental health.

--------------------------------------------------------------
--------------------------------------------------------------

closed 2015-05-15 by amcewen