cheapjack/buildyourown #30: Order spools of PLA for the printing

amcewen, 2015-04-20:
Order 8-12 spools of PLA to provide our source material for prints, in a range of colours.

amcewen, 2015-05-20:
Order placed with Faberdashery.  Will close the issue once it's arrived :-)

LesleyTaker, 2015-05-20:
Ace!!

-- 

Lesley Taker
Curatorial Co-ordinator
FACT (Foundation for Art and Creative Technology)
88 Wood Street
Liverpool, L1 4DQ

Please note, I am only in office until 3pm on a Friday.
tw: @lexy_taylor
t: + 44 (0)151 707 4419
f: + 44 (0)151 707 4445

http://www.fact.co.uk
Admin: +44 (0)151 707 4444
Bookings: +44 (0)871 902 5737

*FACT - Bringing people, art and *
*technology together since 2003*

--------------------------------------------------------------
--------------------------------------------------------------

*Build Your Own: Tools for Sharing4 June – 31 August*

Showcasing four new commissions that span different

models of creative practice, including art, craft, and

creative technology, *Build Your Own* celebrates a new

spirit of generosity through making from your local

street to global maker networks.

--------------------------------------------------------------
--------------------------------------------------------------

amcewen, 2015-05-22:
Arrived today.  Currently sat next to my desk.

closed 2015-05-22 by amcewen