cheapjack/buildyourown #31: Order the additional parts we'll need to assemble the hands

amcewen, 2015-04-20:
Need to order the screws, elastic, velcro, padding, etc. that we'll need to assemble the hands.

Once we've decided on which hands we're building (issue #29) we can then work out what needs to be ordered and place the order.

We should get enough for ~30 hands (at least) as that should cover all we'll make in both Liverpool and Norwich.

amcewen, 2015-05-12:
@DefProc any chance you can work out what parts we need and get them ordered?  Jason donated some bits the other week, which are in his box in the workshop, so that might reduce what we need to order

DefProc, 2015-05-22:
Looking at the enable build list, and a trip round the internet, it looks like our shopping list is:

 * Tensioner Screws (100 will do 20 hands) M3 for small (1 scale) M4 for bigger (e.g. 1.25 scale) M5 for largest. We'll need one of each:
  * 100× M3×20mm Socket Cap stainless screws [ebay](http://www.ebay.co.uk/itm/201271005889?var=500469286553) - £3
  * 100× M4×25mm Socket Cap stainless screws [ebay](http://www.ebay.co.uk/itm/201271005889?var=500469286578) - £4.50
  * 100× M5×25mm Socket Cap stainless screws [ebay](http://www.ebay.co.uk/itm/201271005889?var=500469286620) - £6.50
 * 25m of 50mm wide Velcro One-Wrap (sew on, double sided, "hook and loop" tape). At 1.2m per hand, 25m will do 20 hands. Possible suppliers:
  * [ebay](http://www.ebay.co.uk/itm/Reusable-Hook-Loop-Strapping-Cable-Tie-VELCRO-Brand-One-Wrap-/250789282165?pt=LH_DefaultDomain_3&var=&hash=item3a6434c175) £50
  * [scf supplies](http://www.scfsupplies.co.uk/-One-Wrap-50mm-x-25-Metres/421.htm) - £50
  * [john downes](http://www.johndowns.co.uk/index.php?main_page=product_info&cPath=29_31&products_id=155) - £50
  * [fosseway tapes](http://www.fossewaytapes.co.uk/Self-Adhesive-Velcro-%C2%AE-Brand-Tapes/Self-Adhesive-50mm-2inch-X-25m-VELCRO-%C2%AE-Brand-tape.html) - £55
 * 5mm thick, adhesive backed, closed-cell, firm polyethelyne foam, e.g. Plastizote LD45 (EFP30, Northfoam 170, Plastiozte LD33 (softer) may work as alternatives). Approx 450cm^2 required per hand (40 per 1×2m sheet). Possible suppliers:
  * [easy foam (Simply Foam Product Ltd)](http://www.easyfoam.co.uk/zotefoam-plastazote.php) - £31 for LD33
  * [efoam (Simply Foam Products Ltd)](http://www.efoam.co.uk/closed-cell-polyethylene-foam.php) - £26 for EFP30
 * Spectra/Dyneema/UHDPE braided line >0.5mm diameter, 2.5m per hand. 
  * [ebay](http://www.ebay.co.uk/itm/1000M-100LB-Fishing-Line-Dyneema-Braided-Super-Strong-4-Strands-UK-Delivery-/151497675520) - £16 for 1000m
 * 2mm diameter elastic, "shock" cord, 1.5m per hand. Possible suppliers:
  * [ebay](http://www.ebay.co.uk/itm/ELASTIC-BUNGEE-ROPE-SHOCK-CORD-BLACK-WHITE-RED-BLUE-2mm-3mm-4mm-5mm-6mm-8mm-10mm-/201240391374) - £14.50 /100m
  * [ebay](http://www.ebay.co.uk/itm/ELASTIC-BUNGEE-ROPE-SHOCK-CORD-TIE-DOWN-BLUE-2mm-3mm-4mm-5mm-6mm-8mm-10mm-/3611498834010) - £14.50 /100m, £20 /150m

That puts us around £125 for a set of parts to cover at least 20 hands. 

Any notes, observations, objections? I haven't ordered anything yet.

amcewen, 2015-05-22:
How much braided line do we need?  Jason has donated the reel he used, which is Rovex Depth Finder braid line - 45.4kg (strength, not weight of line :-) 0.5mm, 270m (less however much one hand is)

Otherwise, that looks good, presumably optimising for quick delivery given our timescales...

DefProc, 2015-05-25:
270m minus one hand, will still leave over 100 hands-worth (assuming the 2.5m/hand is accurate).

amcewen, 2015-05-26:
Bah.  Managed to accidentally order the braided line too.  Still, everything has been ordered.  We'll need some more foam, but we can see what the stuff we'd ordered is like before ordering more.

Will close this once everything has arrived.

LesleyTaker, 2015-05-26:
Nice one!!

-- 

Lesley Taker
Curatorial Co-ordinator
FACT (Foundation for Art and Creative Technology)
88 Wood Street
Liverpool, L1 4DQ

Please note, I am only in office until 3pm on a Friday.
tw: @lexy_taylor
t: + 44 (0)151 707 4419
f: + 44 (0)151 707 4445

http://www.fact.co.uk
Admin: +44 (0)151 707 4444
Bookings: +44 (0)871 902 5737

*FACT - Bringing people, art and *
*technology together since 2003*

--------------------------------------------------------------
--------------------------------------------------------------

*Build Your Own: Tools for Sharing4 June – 31 August*

Showcasing four new commissions that span different

models of creative practice, including art, craft, and

creative technology, *Build Your Own* celebrates a new

spirit of generosity through making from your local

street to global maker networks.

--------------------------------------------------------------
--------------------------------------------------------------

DefProc, 2015-05-28:
In an off-by-one error, we will also need some self-tapping screws for palms. Which I need to size.

amcewen, 2015-05-29:
Still waiting on the M4 and M5 screws, but everything else has arrived

LesleyTaker, 2015-06-01:
Know when they'll get here?
L

-- 

Lesley Taker
Curatorial Co-ordinator
FACT (Foundation for Art and Creative Technology)
88 Wood Street
Liverpool, L1 4DQ

Please note, I am only in office until 3pm on a Friday.
tw: @lexy_taylor
t: + 44 (0)151 707 4419
f: + 44 (0)151 707 4445

http://www.fact.co.uk
Admin: +44 (0)151 707 4444
Bookings: +44 (0)871 902 5737

*FACT - Bringing people, art and *
*technology together since 2003*

--------------------------------------------------------------
--------------------------------------------------------------

*Build Your Own: Tools for Sharing4 June – 31 August*

Showcasing four new commissions that span different

models of creative practice, including art, craft, and

creative technology, *Build Your Own* celebrates a new

spirit of generosity through making from your local

street to global maker networks.

--------------------------------------------------------------
--------------------------------------------------------------

cheapjack, 2015-06-01:
Not sure but we will have enough parts and replacement screws to be getting on with aside from nice gel grips that Baylee's dad is getting for us

amcewen, 2015-06-02:
We've got enough to assemble the hands we need for the launch, so moving this to the first assembly day milestone as that's when we'll need the rest of stuff.

amcewen, 2015-06-02:
Reopening this until we have received all the parts we've ordered.

amcewen, 2015-07-30:
Given some of the hands are smaller, I've ordered 20m of 1mm shock cord to make them easier to assemble.

@DefProc Are we happy with the foam we got?  IIRC we didn't buy the full amount as you wanted to check it was suitable.  Is it suitable?  Do we need to buy any more?

DefProc, 2015-08-03:
Yes, I'm happy with the foam. Although it's not a medical grade (there's 
no test for allergic reactions or limit on the ingredients for long term 
use next to skin), it was a good thickness and good firmness, and is 
suitable for this short term, testing, experimentation use.

Yes, we'll need some more. Another 3-4 strips should be plenty.

amcewen, 2015-08-03:
More foam ordered.

marionofnorwich, 2015-09-14:
Hello guys
Is there any kind of review or update on this list now that the project has run in Liverpool?
Marion

cheapjack, 2015-09-14:
We need to do another audit of the non-3d Printed parts that norwich may require to prevent needlessly buying things twice eg. I think we have plenty of velcro and let Marion know about it. We have started a Norwich box in the workshop at DoES for everything we need to send them.

cheapjack, 2015-09-14:
@marionofnorwich we will update the list this week, see my previous comment & email :+1:

marionofnorwich, 2015-09-14:
Thanks @cheapjack . Do you think that there is any mileage in us speaking with Jason? Drew really helpful but doesn't have time to come to Norwich I don't think. Any tips on the functionality would help, though really good talking to Drew in terms of expectations of the Raptor

closed 2015-10-26 by cheapjack