cheapjack/buildyourown #35: Write a blog post with a DesktopProsthetics tag on the DoES website

cheapjack, 2015-04-22:
Something to point anyone at Maker Faire/Reach and other partners and public to information on the project without them having to read everything on github. First post to include the copy from the main README.md file

LesleyTaker, 2015-04-22:
Let us know when this is up, we can use the same copy on our own website, or clone a blog post if that's how you choose to do it!!

cheapjack, 2015-04-28:
We could compile this from the [FAQ](https://github.com/cheapjack/buildyourown/blob/master/FAQ.md) or README will do if someone gives me permissions or whatever...

cheapjack, 2015-04-30:
I've saved a draft of a  blog post if @amcewen or @DefProc could have a quick look and see if Ive missed anything. Obviously need the eventbrite links when setup.

DefProc, 2015-05-09:
This is up now, right?

closed 2015-05-09 by DefProc