cheapjack/buildyourown #37: Organise a date for Front of House training

LesleyTaker, 2015-04-30:
I am getting the team trained in basic 3D Printing using the UPs, through Mark Wright and Jon Couch who has already received training, but it would be better if we could do some specific training for the Ultimakers and your project, if that seems do-able? Would you guys like to recommend some dates one, or more of you might be able to do? Jon C will facilitate it and support you and we could use the Media Lab here, or send them down to DoES- whatever you think.

cheapjack, 2015-04-30:
Once we know delivery times of the Ultimakers we can set this date; would make sense if they came to DoES as we will be testing and preprinting stuff for the show great idea. Will have to get ourselves up to speed also!

LesleyTaker, 2015-04-30:
Sounds good to me, just let me know!! I am arranging some general 3D
training, probably on 30th May because that's just after our next volunteer
intake and general FACT training, but can move this earlier if necessary?

-- 

Lesley Taker
Curatorial Co-ordinator
FACT (Foundation for Art and Creative Technology)
88 Wood Street
Liverpool, L1 4DQ

Please note, I am only in office until 3pm on a Friday.
tw: @lexy_taylor
t: + 44 (0)151 707 4419
f: + 44 (0)151 707 4445

http://www.fact.co.uk
Admin: +44 (0)151 707 4444
Bookings: +44 (0)871 902 5737

*FACT - Bringing people, art and *
*technology together since 2003*

--------------------------------------------------------------
--------------------------------------------------------------

*Group Therapy: Mental Distress in a Digital Age5 March – 17 May*

Originating from FACT’s extensive work within

mental health and wellbeing, the exhibition

explores the complex relationship between

technology, society, and mental health.

--------------------------------------------------------------
--------------------------------------------------------------

amcewen, 2015-05-02:
Tweaked the title as that's what we want to sort out and arranging a date is just the first step on that.  Currently stalled on issue #2 as @cheapjack said (but now hopefully when that issue gets closed we'll spot that this one is freed up)

cheapjack, 2015-05-06:
Ok Im going to list some possible dates for training GA's, prob best in week its most likely to be set up ready to go 

Monday 1st June
* [x] @cheapjack
* [ ] @amcewen 
* [ ] @DefProc 

Tuesday 2nd June
* [x] @cheapjack
* [ ] @amcewen 
* [ ] @DefProc 

Wednesday 3rd June
* [ ] @cheapjack
* [ ] @amcewen 
* [ ] @DefProc

LesleyTaker, 2015-05-13:
Just to let you know, Front of House have said their preference would be Wed 3 June!

LesleyTaker, 2015-05-26:
Hey all, what date did we decide for this? Really need to book it in with the volunteers etc 
Thanks!!!

cheapjack, 2015-05-26:
Wednesday 3rd June will be fine for me and @DefProc (see above) and we wont need @amcewen for that session, just update us on the time. As we havent fixed the date on getting the Ultimakers installed and running in the space yet, Id suggest a time late afternoon on the 3rd June. @LesleyTaker could you let us know when that week you think space would be clear, safe enough and powered to setup and start printing? Im thinking Wednesday may be best; we dont want to install them too soon and not be able to print or interrupt prints so power will have to be fixed for us. Also gives us maximum time for any overnight prints we might need.

LesleyTaker, 2015-05-26:
Late on Tuesday (2nd) or Wednesday (3rd) morning would be fine to put them
in.

Do you want to do the training with them all installed or do training maybe
in another space then install them? Because that obviously affects the
timing of placing them etc. I assume it would be best to do it in situ?
With this in mind, let me know what time you'd to do the training. The
techs break for an hour for lunch at 1pm, and finish (theoretically) at
5.30pm, just FYI. But there shouldn't be that much to do on the Wednesday
anyway.

Will you guys have other (obviously smaller) hands ready so that people can
see / use them on the opening night? Just aware we will only like have one
from the Ultimakers...

-- 

Lesley Taker
Curatorial Co-ordinator
FACT (Foundation for Art and Creative Technology)
88 Wood Street
Liverpool, L1 4DQ

Please note, I am only in office until 3pm on a Friday.
tw: @lexy_taylor
t: + 44 (0)151 707 4419
f: + 44 (0)151 707 4445

http://www.fact.co.uk
Admin: +44 (0)151 707 4444
Bookings: +44 (0)871 902 5737

*FACT - Bringing people, art and *
*technology together since 2003*

--------------------------------------------------------------
--------------------------------------------------------------

*Build Your Own: Tools for Sharing4 June – 31 August*

Showcasing four new commissions that span different

models of creative practice, including art, craft, and

creative technology, *Build Your Own* celebrates a new

spirit of generosity through making from your local

street to global maker networks.

--------------------------------------------------------------
--------------------------------------------------------------

cheapjack, 2015-05-26:
No we need to do the training with them installed as that' s where they will be using them and will need to go through our system for keeping track of parts etc. Maybe 3pm would be ok: @DefProc? 

We are planning on printing one for the opening night aside from the vitrine display ones. Most likely one. It may be printed on the Ultimaker or the UP

LesleyTaker, 2015-05-26:
3pm sounds good, will wait for @DefProc sign off though. Is there any way
we can have one or two more printed in advance? Purely because people will
definitely want to pick them up, see how they feel etc, and it will be
super busy on the opening night.

-- 

Lesley Taker
Curatorial Co-ordinator
FACT (Foundation for Art and Creative Technology)
88 Wood Street
Liverpool, L1 4DQ

Please note, I am only in office until 3pm on a Friday.
tw: @lexy_taylor
t: + 44 (0)151 707 4419
f: + 44 (0)151 707 4445

http://www.fact.co.uk
Admin: +44 (0)151 707 4444
Bookings: +44 (0)871 902 5737

*FACT - Bringing people, art and *
*technology together since 2003*

--------------------------------------------------------------
--------------------------------------------------------------

*Build Your Own: Tools for Sharing4 June – 31 August*

Showcasing four new commissions that span different

models of creative practice, including art, craft, and

creative technology, *Build Your Own* celebrates a new

spirit of generosity through making from your local

street to global maker networks.

--------------------------------------------------------------
--------------------------------------------------------------

amcewen, 2015-06-02:
I've generated the files for another demonstration hand - checked into git as cad/hands/demonstration-85.

We should kick off printing that hand out on Thursday so the printers have something to run with during the first couple of days.  It will also let front-of-house run through the procedure for a print correctly too.

We can then assemble it during the workshop on Saturday (it'll help explain how things go together) and then be left for the public to try on.

DefProc, 2015-06-05:
train'd

closed 2015-06-05 by DefProc