cheapjack/buildyourown #39: Document the measuring up -> files -> prints -> hand workflow

amcewen, 2015-05-07:
Need to document how we're going to track different people's measurements, preferences (colours, etc.) and details, and then how we match that to printed out parts, etc.

Also needs to include how we loop the e-NABLE EnableMatcher into the process.

LesleyTaker, 2015-05-13:
Let me know what you need from me around ensuring people's pieces go in the correct place. We are getting a lot of emotive and very positive attention from families for the project, so we need to make absolutely sure we're not losing fingers! ;)

cheapjack, 2015-05-20:
We could do with knowing details of the pegboard storage options we talked about with the Bloors
It may make sense to have paper part bags to keep everything in and some labels on hand to stick onto the boxes

cheapjack, 2015-05-27:
It's more work, but be useful to document this as a handout for the Gallery Guides to answer workflow FAQ's save us doing that in person to make best use of our support time through the show.

cheapjack, 2015-05-30:
I've started an assembly manual and am gradually adding images and notes to it for Gallery Guides to refer to for our Assembly day. I'm dumping the images here and then shifting them into the manual

[image] 
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]