cheapjack/buildyourown #41: We should flip the proximal upside down to print

amcewen, 2015-05-15:
If we print the proximals upside down (compared to how they are at present) then there'll be less support material to remove.

amcewen, 2015-05-15:
I'm doing a test print with it now - I hadn't realised until this test that there's a raised bit on the top of the proximal (the power-cut we had in the workshop earlier meant the top bit hadn't finished printing :-) so we'll see how that comes out.  Might still be less fiddly to remove the support material, even if there's a bit more of it printed

DefProc, 2015-05-22:
Based on extra top bit on the proximal, and because the Ultimakers *should* print without support material, I'll close this for now.

closed 2015-05-22 by DefProc