cheapjack/buildyourown #43: NERF Firing Range at BYO Opening

LesleyTaker, 2015-05-20:
Just spoke to everyone and they love the idea of doing this!!! We will advertise it as running from 5-6pm if that's ok with you? And if it's super popular we will just keep it running through the exhibition? I will try and get some of our volunteers to help man this but would obviously be great for some people who know about it more to man it as well!!
L

cheapjack, 2015-05-20:
Just so you know at Maker Faire it was not a question of 'if' it's super popular; people like shooting things! ;) Brett at least has volunteered to man it on the DoES side but the more FACT volunteers the better

If it's raining we can't do it outside is the only thing to note :) Also assuming you mean run it through the course of the evening not the whole exhibition period :+1:

LesleyTaker, 2015-05-20:
Definitely just for the launch! ;)
Until like 7.30pm maybe?
I'll try and get some more vols on side!!

-- 

Lesley Taker
Curatorial Co-ordinator
FACT (Foundation for Art and Creative Technology)
88 Wood Street
Liverpool, L1 4DQ

Please note, I am only in office until 3pm on a Friday.
tw: @lexy_taylor
t: + 44 (0)151 707 4419
f: + 44 (0)151 707 4445

http://www.fact.co.uk
Admin: +44 (0)151 707 4444
Bookings: +44 (0)871 902 5737

*FACT - Bringing people, art and *
*technology together since 2003*

--------------------------------------------------------------
--------------------------------------------------------------

*Build Your Own: Tools for Sharing4 June – 31 August*

Showcasing four new commissions that span different

models of creative practice, including art, craft, and

creative technology, *Build Your Own* celebrates a new

spirit of generosity through making from your local

street to global maker networks.

--------------------------------------------------------------
--------------------------------------------------------------

DefProc, 2015-05-22:
If there's a space inside, then no problem. It all fits in a 3×4.5m gazebo, open on the short end. It's not designed to be put up outside (computers, electronics and cardboard all without any weatherproofing, and directly on the floor), so it's highly dependent on the weather — especially as it has to be ready to go to Liverpool MakeFest after. 

Is there any space in the main area? Is it worth me coming in to see if there's a good space for it?

LesleyTaker, 2015-05-26:
It would have to be outside, there's definitely won't be enough space in
the foyer, I'm afraid :(
So do we just play it by ear with the weather, or is there anything we can
do to help make it more weather proof? or should I go back to the 'opening
activity' drawing board?

-- 

Lesley Taker
Curatorial Co-ordinator
FACT (Foundation for Art and Creative Technology)
88 Wood Street
Liverpool, L1 4DQ

Please note, I am only in office until 3pm on a Friday.
tw: @lexy_taylor
t: + 44 (0)151 707 4419
f: + 44 (0)151 707 4445

http://www.fact.co.uk
Admin: +44 (0)151 707 4444
Bookings: +44 (0)871 902 5737

*FACT - Bringing people, art and *
*technology together since 2003*

--------------------------------------------------------------
--------------------------------------------------------------

*Build Your Own: Tools for Sharing4 June – 31 August*

Showcasing four new commissions that span different

models of creative practice, including art, craft, and

creative technology, *Build Your Own* celebrates a new

spirit of generosity through making from your local

street to global maker networks.

--------------------------------------------------------------
--------------------------------------------------------------

DefProc, 2015-05-26:
If you need a definate, yes have a look for another exhibitor. I won't have any time to weather proof it before the opening, sorry.

LesleyTaker, 2015-05-28:
Ok, just spoke to everyone. We can have it in The Box- how long does it take to put up / pack down? 
It would be great if this is a plan B because it would be SO SO awesome to have it outside if it's a nice day!!!

DefProc, 2015-06-05:
Just needs the boxes and gazebo collecting from the delivery bay in FACT.

DefProc, 2015-06-08:
All collected and returned to DoES

closed 2015-06-08 by DefProc

LesleyTaker, 2015-06-11:
THANK YOU SO MUCH! Was a total hit. <3