cheapjack/buildyourown #44: Complete the Sub-Plinth storyboard Vinyl Artwork and design

cheapjack, 2015-05-20:
Completed Artwork will live here once Ive finished it
https://github.com/cheapjack/buildyourown/tree/master/Exhibition/FACT/interpretation

Can Lesley or Alma send me the Build Your Own Font? ta

cheapjack, 2015-05-20:
Sorry also can you let me know what colour we are working to for the story table/plinth and the subplinths and also our area of the show? originally was green, that still the case? so I know what vinyl colour to go for thanks!

LesleyTaker, 2015-05-20:
Green!!

-- 

Lesley Taker
Curatorial Co-ordinator
FACT (Foundation for Art and Creative Technology)
88 Wood Street
Liverpool, L1 4DQ

Please note, I am only in office until 3pm on a Friday.
tw: @lexy_taylor
t: + 44 (0)151 707 4419
f: + 44 (0)151 707 4445

http://www.fact.co.uk
Admin: +44 (0)151 707 4444
Bookings: +44 (0)871 902 5737

*FACT - Bringing people, art and *
*technology together since 2003*

--------------------------------------------------------------
--------------------------------------------------------------

*Build Your Own: Tools for Sharing4 June – 31 August*

Showcasing four new commissions that span different

models of creative practice, including art, craft, and

creative technology, *Build Your Own* celebrates a new

spirit of generosity through making from your local

street to global maker networks.

--------------------------------------------------------------
--------------------------------------------------------------

cheapjack, 2015-05-26:
Im suddenly worried about ledgibility of the text on green was planning on white vinyl but is it too late to have the table green as planned but the sub plinths painted white?

LesleyTaker, 2015-05-26:
It'd look a bit weird as everything else is dyed, there is no paint at all.
White on green will look beautiful!!

-- 

Lesley Taker
Curatorial Co-ordinator
FACT (Foundation for Art and Creative Technology)
88 Wood Street
Liverpool, L1 4DQ

Please note, I am only in office until 3pm on a Friday.
tw: @lexy_taylor
t: + 44 (0)151 707 4419
f: + 44 (0)151 707 4445

http://www.fact.co.uk
Admin: +44 (0)151 707 4444
Bookings: +44 (0)871 902 5737

*FACT - Bringing people, art and *
*technology together since 2003*

--------------------------------------------------------------
--------------------------------------------------------------

*Build Your Own: Tools for Sharing4 June – 31 August*

Showcasing four new commissions that span different

models of creative practice, including art, craft, and

creative technology, *Build Your Own* celebrates a new

spirit of generosity through making from your local

street to global maker networks.

--------------------------------------------------------------
--------------------------------------------------------------

cheapjack, 2015-05-26:
OK see what you think [on the pdf here](https://github.com/cheapjack/buildyourown/blob/master/Exhibition/FACT/interpretation/Vitrines.pdf)

LesleyTaker, 2015-05-26:
I really like it!! I would have the separate actions in each section bullet pointed though, to make it obvious they're separate steps in the same stage. Looks beautiful though!!

I was just thinking- do you still want a secondary, accessible iPad in the space so people can access the handomatic or we display the wiki?

Sent from somewhere else.

almataki, 2015-05-27:
Looks really good guys! great job. Lesley has a point about separating the teps with bullet poitns or numbers.

cheapjack, 2015-05-31:
Cut, prepped ready for install

closed 2015-05-31 by cheapjack