cheapjack/buildyourown #46: Confirm the install date and complete the Install notes

cheapjack, 2015-05-26:
I've started a [list of things we need for the install](https://github.com/cheapjack/buildyourown/blob/master/Exhibition/FACT/InstallNotes.md) and suggested we set the install of printers on Wednesday 3rd June Morning to ensure space clear and ready with power especially to avoid print interruptions and allow us to print right up to the deadline

It will make sense to sort the vitrines on Tuesday and maybe the storage also. Im available Tuesday to look at that.

LesleyTaker, 2015-05-26:
Just so you know, have ordered all vitrines and they will be there sometime
before midday on the Wednesday (was the earliest I could get them I'm
afraid!!)
L

-- 

Lesley Taker
Curatorial Co-ordinator
FACT (Foundation for Art and Creative Technology)
88 Wood Street
Liverpool, L1 4DQ

Please note, I am only in office until 3pm on a Friday.
tw: @lexy_taylor
t: + 44 (0)151 707 4419
f: + 44 (0)151 707 4445

http://www.fact.co.uk
Admin: +44 (0)151 707 4444
Bookings: +44 (0)871 902 5737

*FACT - Bringing people, art and *
*technology together since 2003*

--------------------------------------------------------------
--------------------------------------------------------------

*Build Your Own: Tools for Sharing4 June – 31 August*

Showcasing four new commissions that span different

models of creative practice, including art, craft, and

creative technology, *Build Your Own* celebrates a new

spirit of generosity through making from your local

street to global maker networks.

--------------------------------------------------------------
--------------------------------------------------------------

cheapjack, 2015-05-26:
Ok lets say Wednesday then: You got a link to what you've ordered? Im sure it could be done all on the Wednesday but just gives us more time if we need anything extra doing.

LesleyTaker, 2015-05-26:
I haven't got a link because they're custom made (clear acrylic vitrines
with threes sides and a top, no base) from these guys:
http://www.bayplastics.co.uk/

I ordered *4x *420mm x 500mm x 100mm H

Do we need anything to cover the printers?

-- 

Lesley Taker
Curatorial Co-ordinator
FACT (Foundation for Art and Creative Technology)
88 Wood Street
Liverpool, L1 4DQ

Please note, I am only in office until 3pm on a Friday.
tw: @lexy_taylor
t: + 44 (0)151 707 4419
f: + 44 (0)151 707 4445

http://www.fact.co.uk
Admin: +44 (0)151 707 4444
Bookings: +44 (0)871 902 5737

*FACT - Bringing people, art and *
*technology together since 2003*

--------------------------------------------------------------
--------------------------------------------------------------

*Build Your Own: Tools for Sharing4 June – 31 August*

Showcasing four new commissions that span different

models of creative practice, including art, craft, and

creative technology, *Build Your Own* celebrates a new

spirit of generosity through making from your local

street to global maker networks.

--------------------------------------------------------------
--------------------------------------------------------------

cheapjack, 2015-05-26:
Ok is there a subplinth that would be ready Wednesday morning?

No need to cover printers as far as i know

LesleyTaker, 2015-05-26:
The suplinths will all be ready- your space should be completely finished
by then to be honest.

-- 

Lesley Taker
Curatorial Co-ordinator
FACT (Foundation for Art and Creative Technology)
88 Wood Street
Liverpool, L1 4DQ

Please note, I am only in office until 3pm on a Friday.
tw: @lexy_taylor
t: + 44 (0)151 707 4419
f: + 44 (0)151 707 4445

http://www.fact.co.uk
Admin: +44 (0)151 707 4444
Bookings: +44 (0)871 902 5737

*FACT - Bringing people, art and *
*technology together since 2003*

--------------------------------------------------------------
--------------------------------------------------------------

*Build Your Own: Tools for Sharing4 June – 31 August*

Showcasing four new commissions that span different

models of creative practice, including art, craft, and

creative technology, *Build Your Own* celebrates a new

spirit of generosity through making from your local

street to global maker networks.

--------------------------------------------------------------
--------------------------------------------------------------

cheapjack, 2015-05-31:
Vinyl is cut and prepped ready to apply. Time willing will start this on Tuesday afternoon so me and @defproc can concentrate on printer setup and workflow on Wednesday in advance of staff training.

closed 2015-05-31 by cheapjack