cheapjack/buildyourown #54: Prep talk for the artists talk on Friday

amcewen, 2015-06-02:
Finish off slides & any other prep before Friday's artists talk.

amcewen, 2015-06-02:
@lesleytaker are there any guidelines, etc. for the artists talks?  How long?  I'm assuming there'll be a projector?

LesleyTaker, 2015-06-02:
It's in the Box, am sure Amy will be chasing you for presentations tomorrow
and Thursday, if you want to do one, show images / film etc all of that is
fine. They are 15 mins max. for each artist / collective. The drive behind
doing them was just so everyone could talk about their project, how it fits
into the thematic of BYO: Tools For Sharing and so we could show how it
links traditional craft to technological craft- basically the same theory
as behind the show.

-- 

Lesley Taker
Curatorial Co-ordinator
FACT (Foundation for Art and Creative Technology)
88 Wood Street
Liverpool, L1 4DQ

Please note, I am only in office until 3pm on a Friday.
tw: @lexy_taylor
t: + 44 (0)151 707 4419
f: + 44 (0)151 707 4445

http://www.fact.co.uk
Admin: +44 (0)151 707 4444
Bookings: +44 (0)871 902 5737

*FACT - Bringing people, art and *
*technology together since 2003*

--------------------------------------------------------------
--------------------------------------------------------------

*Build Your Own: Tools for Sharing4 June – 31 August*

Showcasing four new commissions that span different

models of creative practice, including art, craft, and

creative technology, *Build Your Own* celebrates a new

spirit of generosity through making from your local

street to global maker networks.

--------------------------------------------------------------
--------------------------------------------------------------

-- 
 

*The contents of this email and any attachments are intended solely for the 
original recipient(s). If you have received it in error please contact the 
sender immediately by returning the e-mail or by telephoning a number 
contained in the body of the e-mail, then please delete the e-mail 
immediately without disclosing its contents. No responsibility is accepted 
for loss or damage arising from viruses or changes made to this message 
after it was sent. The views contained in this email are those of the 
author and not necessarily those of the authors employer or service 
provider.*

DefProc, 2015-06-05:
Present'd

closed 2015-06-05 by DefProc