cheapjack/buildyourown #64: Empty seeming SD Cards [urgent]

LesleyTaker, 2015-06-11:
Hey, sorry if I've missed something but the printers are currently not printing anything, looks like they've completed everything that was on the SD cards. Sorry if this is something you told Mark / Jon - both of them are off today (Mark's off for ages) and I can't find anything in the other issues / guide other than a print queue looking sparse:
Are there any other cards anywhere, or do we need to queue stuff/ do you guys need to? 
Thanks
!!!

cheapjack, 2015-06-11:
Hi Lesley 
The outcome of Saturday was that many children participation turned out not to have working wrist joints to make the Raptor reloaded the best design for their situation. So we are just trying to decide the best course of action. We will be contacting the families to explain this.

We populated the disks on the printer for this week to print a hand that was not affected by this but I see we must have completed this print by now.

We photographed and made measurements and notes for each participant on the workshop and in the case of Harrisson (a young lad all the way from Devon), he was happy experimenting with the raptor reloaded using his elbow joint, and took one home, which although not ideal did make him feel part of the project and they understood its limitations in relation to his situation. We need to work back through the notes that we made with Drew and work out the best responses for everyone which we've not had time to do yet. Options are so far;

1 Still print out Raptor Reloaded designs for everyone as planned and let families know that this model may not be the best design for them but is a starting point in a co-design process that we can build on in the subsequent workshops and at DoES as the interest group around the project grows

2 Print out alternative designs : they exist but we have no experience building or assembling or fitting them and unclear whether we would be able to make them a good fit for the participants remotely: so we could end up with prosthetics as problematic as a raptor for someone with no working wrist.

Meantime we should generate some more raptor files to cover the next few days: but I think option 1 now seems our best course of action. What do @amcewen @DefProc think? We talked about it yesterday and decided to delay decision and getting back to families to next week not realising prints would be complete by now. For time being/weekend we could print out another small demonstrator hand and Thwack hammer: so I can set one of those off tomorrow and explain to whoever is on.

LesleyTaker, 2015-06-11:
Oh god! Ok. Didn't know this was the case.

Do we really want to print out hands that these kids won't be able to use?
if we have their measurements how hard is it to adapt them to the other
models?

Maybe we can reach out (seriously no pun intended) to someone at Reach, or
someone who has more experience with a diverse range of hands to assist in
assembling etc? There's a bit of money left for their travel if we want to
go this way? It might be a good idea because then you guys are also
equipped to deal with a variety of hands if people come to use the
Ultimaker for their prosthetics?

This might be a rubbish idea, just thinking out loud.
Do you want me to come down for a chat tomorrow or something?

And nice one of the replacement demonstrator and the hammer, @cheapjack -
good call!

-- 

Lesley Taker
Curatorial Co-ordinator
FACT (Foundation for Art and Creative Technology)
88 Wood Street
Liverpool, L1 4DQ

Please note, I am only in office until 3pm on a Friday.
tw: @lexy_taylor
t: + 44 (0)151 707 4419
f: + 44 (0)151 707 4445

http://www.fact.co.uk
Admin: +44 (0)151 707 4444
Bookings: +44 (0)871 902 5737

*FACT - Bringing people, art and *
*technology together since 2003*

--------------------------------------------------------------
--------------------------------------------------------------

*Build Your Own: Tools for Sharing4 June – 31 August*

Showcasing four new commissions that span different

models of creative practice, including art, craft, and

creative technology, *Build Your Own* celebrates a new

spirit of generosity through making from your local

street to global maker networks.

--------------------------------------------------------------
--------------------------------------------------------------

-- 
 

*The contents of this email and any attachments are intended solely for the 
original recipient(s). If you have received it in error please contact the 
sender immediately by returning the e-mail or by telephoning a number 
contained in the body of the e-mail, then please delete the e-mail 
immediately without disclosing its contents. No responsibility is accepted 
for loss or damage arising from viruses or changes made to this message 
after it was sent. The views contained in this email are those of the 
author and not necessarily those of the authors employer or service 
provider.*

cheapjack, 2015-06-11:
We basically don't know how hard it would be to use the RIT Arm what the cad files are like or design it for the kids we met on Saturday. We will look into it for sure but meantime example hands could cover us temporarily so the exhibition is active. It's worth trying one out we've just not built in the time to do that yet.

Appreciate the suggestion Lesley but it ups the scope of the project a fair bit researching prosthetic 'experts' at this stage and working with Reach like that I think plus it was always a starting point what we were doing: I think the e-nable people would be more appropriate than reach: and Drew already helping: he has alot of experience and enthusiasm and is already helping us with this. If we could pay for Drew's travel to attend the assembly workshops and make the prosthetic hack day about this issue that's a good idea: though this is an additional cost to our original budget I guess.

We can get some elements printing and then come up with a considered solution. These were always going to be prototypes and it's a process so don't think we have to panic into a solution: I know you want the exhibition visibly printing but there is always down time when you prototyping things.

That said we do want to keep everyone informed and will draft something so all participants are aware of what we're doing next week.

I could pop in tomorrow morning and set off another demo hand and have a look over things.

cheapjack, 2015-06-12:
@defproc kindly generated some plates for a raptor gasp-law which will keep the prints active until Tuesday. I've suggested just running this on one printer on the right and leave plate 5 of guts-squid on display as it looks nice and explains process nicely

cheapjack, 2015-06-17:
We've populated the SD cards on the Ultimakers with ```Gin-Timers``` and ```Leaf-Keeps``` for those families who are best suited. We will continue to generate files for Raptor hands and transferring to SD card with a view to fitting and modding them at the Assembly workshops in August and the Prosthetic Hackday in July.

closed 2015-06-17 by cheapjack