cheapjack/buildyourown #66: Generate a new batch of files for delivery to FACT

cheapjack, 2015-06-29:
All files currently printed need 2 or more hand files for @cheapjack to deliver to FACT tomorrow 30/06/15 if at all possible @DefProc ;) ta

LesleyTaker, 2015-06-29:
Thanks guys! Told Wash and Jon to expect you tomorrow.

cheapjack, 2015-06-30:
Will get to this when we can Lesley if not today at some point this week. @DefProc when you've stopped seeing NERF pellets in your sleep I'm happy to deliver at FACT ;)

LesleyTaker, 2015-06-30:
Hey @cheapjack @DefProc would be good as soon as poss!! They've been quiet for a couple of days apparently, I only just got told yesterday.

cheapjack, 2015-07-01:
As soon as @DefProc has generated I will pop in and pop on the machines.
We heard from you that the queue was complete late Friday afternoon but in middle of LPoolMakeFest prep. so we have had no chance to work on it yet so bear with us ta :)

cheapjack, 2015-07-13:
More hands please @DefProc :) will drop them in to FACT if you can generate some for me :) I really should learn how to...

amcewen, 2015-07-14:
Have we got more hands to print @DefProc?  Are they all done now?  Do we need to generate some arm prints to test out?  FACT would really like their printers to be doing something (as would we, I'm sure)

cheapjack, 2015-07-17:
Enough Files delivered for a while now at least until 1st assembly day. 10 hands in queue at present with near 5 complete.

closed 2015-07-17 by cheapjack