cheapjack/buildyourown #69: Nozzle jammed on printer on the right

muff7, 2015-07-01:
Seems to not want to extrude any filament.

Although the feeder is still moving the filament around when it comes to the print head it skips as though it can't push through. Is this something you trouble shoot or is it down to be to jamb a needle up there?!

DefProc, 2015-07-01:
My first approach would be to do a "cold pull" clean on the nozzle: 
http://support.3dverkstan.se/article/10-the

It solved under extrusion and a blocked nozzle on our workshop model.

Are you happy to do it, or do you need me to come in?

cheapjack, 2015-07-17:
Saw @muff7 in FACT other day, I was going to email him but seemed to have lost his mail so will mail Joan instead with the [Cold Pull Link](http://support.3dverkstan.se/article/10-the)

closed 2015-07-17 by cheapjack

cheapjack, 2015-07-17:
Durr I've confused @muff7 with John in the gallery. @muff7 is quite clearly acronymed-ish from Mark Murphy :)