cheapjack/buildyourown #70: Support for Norwich

marionofnorwich, 2015-07-09:
Just trying out how to communicate via Github. Just to say that we have printed a hand down here but have no assembly materials as yet. Hoping to sort that out soon

almataki, 2015-07-09:
@marionofnorwich well done for printing the first hand!!

marionofnorwich, 2015-07-09:
Thanks Alma. Once of our members has an Ultimaker and kindly printed two off actually, one small, one large so we at least have something to play with!

amcewen, 2015-07-14:
Cool that you've got some hands printed to try out.

I gave @marionofnorwich some padding, velcro, fishing line and elastic when she called in on Saturday, so hopefully they'll be up and running on getting the hands assembled now.

closed 2015-07-14 by amcewen

marionofnorwich, 2015-07-14:
Yes, we had a go at assembling last night but the elastic is too thick to go through the holes on the small size hand we printed so need to look for thinner elastic. The larger hand we need more parts for as was only part printed

amcewen, 2015-07-15:
If you need pointers on where to get bits, there's a list of links for where we got stuff in issue #31

marionofnorwich, 2015-07-15:
Yes, thanks Adrian

DefProc, 2015-07-18:
We got 2mm diameter elastic, but I actually like 1mm better — especially for the smaller builds.

marionofnorwich, 2015-07-18:
Yes I think it would be much better for us too. Was it easy to get hold of?

DefProc, 2015-07-18:
I got our 2mm from ebay in a roll. 1mm was available too.

marionofnorwich, 2015-07-19:
Thanks. I have mocked up this flyer to recruit people to the first workshop, though it doesn't mention dates. Thought if I can get people to get in touch i can explain what's needed and when it is. Or could put more info on a second page/reverse. What do you think?
[image]

LesleyTaker, 2015-07-20:
Hey Marion
,
Just a few quick things!

   - FACT Liverpool should be FACT (Foundation for Art and Creative
   Technology) in the credit bit
   - No "the" before Crafts Council
   - Liverpool DoES should be DoESLiverpool (one word)
   - *Build Your Own: Tools for Sharing* (needs second part of title) and
   should be italicised
   - Probably good to mentioned Enable in there somewhere?

Also, I would change the bit where it says the Hackspace and DoES have
co-produced the exhibition, maybe put it in another location and have '*Desktop
Prosthetics* facilitated by' / 'co-ordinated by' or even credit as artists.
Up to you guys really!

Best,
L

-- 

Lesley Taker
Curatorial Co-ordinator
FACT (Foundation for Art and Creative Technology)
88 Wood Street
Liverpool, L1 4DQ

Please note, I am only in office until 3pm on a Friday.
tw: @lexy_taylor
t: + 44 (0)151 707 4419
f: + 44 (0)151 707 4445

http://www.fact.co.uk
Admin: +44 (0)151 707 4444
Bookings: +44 (0)871 902 5737

*FACT - Bringing people, art and *
*technology together since 2003*

--------------------------------------------------------------
--------------------------------------------------------------

*Build Your Own: Tools for Sharing4 June – 31 August*

Showcasing four new commissions that span different

models of creative practice, including art, craft, and

creative technology, *Build Your Own* celebrates a new

spirit of generosity through making from your local

street to global maker networks.

--------------------------------------------------------------
--------------------------------------------------------------

-- 
 

*The contents of this email and any attachments are intended solely for the 
original recipient(s). If you have received it in error please contact the 
sender immediately by returning the e-mail or by telephoning a number 
contained in the body of the e-mail, then please delete the e-mail 
immediately without disclosing its contents. No responsibility is accepted 
for loss or damage arising from viruses or changes made to this message 
after it was sent. The views contained in this email are those of the 
author and not necessarily those of the authors employer or service 
provider.*

marionofnorwich, 2015-07-20:
OK, thanks Lesley, I will change where I can. Obviously limited space and the main point is to get some interest from the people that we need to get signed up. I will amend and repost

Sorry I missed you when I came to Liverpool last weekend, it was a snatched opportunity but useful to speak to Joan and check in with Adrian

marionofnorwich, 2015-07-20:
[image]

marionofnorwich, 2015-07-20:
Latest version
[image]

almataki, 2015-07-21:
I think this is good to go! looks good to me.

marionofnorwich, 2015-07-21:
Yes, I think so too. Will work with Amanda in next few weeks to work up a
press release but this will enable us to get going with recruitment
Thanks
Marion

Marion Catlin
3 Trowse House Cottages
Bracondale Millgate
Norwich NR1 2EQ

Telephone 01603 765353
Mobile 07946 261651

marionofnorwich, 2015-07-21:
Just agreed the loan of an Ultimaker for the summer hols from Create Education so that we can have a play!

marionofnorwich, 2015-07-22:
How long did you guys run your measuring workshop for?

cheapjack, 2015-07-23:
We ran it 12-4.30 you can see our timing activity plan here

https://github.com/cheapjack/buildyourown/blob/master/events/MeasuringUpWorkshop.md

marionofnorwich, 2015-07-23:
Thanks Ross, that seems about right. We are thinking 11-3pm as the museum closes at 4.30pm so that will give us time for overrun and clearup. Does that sound right to you? I just hpe it will also give people enough time to travel but as you did two sessions, I guess we can cover all travel considerations that way. We have our first candidate - actually someone I know. What I didn't know was that in the last few months he has suffered a neural problem which means he can't use his hand, and he is a guitarist. He is making the most of it but quite distressed as he can't play so it could be a good mission to get him playing. members are enthusiastic generally and Ultimaker are lending us a machine for the holiday arriving tomorrow. So it is moving forward. We also put in a funding bid for a lasercutter for the space so may be back for more advice if we are successful!