cheapjack/buildyourown #71: Ther's no visibility of the print queue

amcewen, 2015-07-14:
It shouldn't be in github, obviously, as that's public, but I don't seem to have any record of the details we collected at the measuring up day.

That means there's no way for me to help out with filling the print queue, or have an overview of where we're up to with the print queue.

cheapjack, 2015-07-14:
Sorry will email you the link to the Google drive document. Will chase @defproc fort new hands

cheapjack, 2015-07-17:
@amcewen now has the drive link, there is a [public print queue here](https://github.com/cheapjack/buildyourown/blob/master/cad/BuildChecklist.md), with just the `hipster-gen` code-names to monitor build process. Last time I looked `gushes-eddy` was 1 plate off completion

closed 2015-07-17 by cheapjack