cheapjack/buildyourown #75: Start reprint queue at FACT

cheapjack, 2015-08-17:
Start reprint queue at FACT while closed if possible
 and check with @amcewen to avoid duplicates

amcewen, 2015-08-17:
I'm using the "in the wild" label to track issues that people have with their hands/wings once they start using them.

amcewen, 2015-08-17:
Duplicate-wise, as long as you don't kick off any of the prints in the DoES Liverpool queue section of https://github.com/cheapjack/buildyourown/blob/master/cad/BuildChecklist.md, we'll be okay.

cheapjack, 2015-08-18:
John and i Started the gin-trite plates printing at FACT today. Below is the rest of the reprints required at FACT I had to give Mark an updated reprints folder `reprintsok` 

Had to restart a print today so we have been tuning the prints in DoES up to 215 degreesNozzle temperature and 80% speed once the print had started. You do this by accessing the Tune menu once a print had started. I set 2 prints off with these settings and they seemed fine. If we can set a print off end of the day for overnight that would be great

Below is the list of things to try print at FACT with colours. We will use dark blue instead of purple.

**Plates required for reprints:**
`undies-manual` dark blue palm / yellow gauntlet
`beds-troy` silver palm / purple/darkblue gauntlet
`tuck-guts` red palm / yellow gauntlet
`gin-trite` dark blue / yellow

If we push we might get them done for Saturday

cheapjack, 2015-08-19:
I've looked through the bookings again for Sat 22nd and everyone except the `gin-trite` reprint are attending so we need to focus on reprinting the following at FACT best we can:  
`undies-manual` 
`beds-troy`
`tuck-guts`

cheapjack, 2015-08-20:
Stopped in FACT today
Washington & John & Mark going best they can and looks like we should at least get
`beds-troy`
`tuck-guts`
printed for Saturday. I think I will need to let the `undies-manual` family know their print may not be ready

cheapjack, 2015-08-21:
I've just heard from the family of the `beds-troy` hand last night they can no longer make it up from Devon tomorrow now so can we please switch one of the printers to undies-manual? 

(Sorry trying to prioritise which prints which on which printer with families changing plans and reprint queues has become a nitemare!)

`undies-manual` in the `reprintsok` folder i gave to mark in yellow and grey. Please set off Grey-3-undies-manual.gcode for today asap and we may get it done by end of the day and try get Grey-4-undies-manual.gcode set off overnight: I could come in late if and set it off if really necessary Dave/security could let me in? 

It would mean that we might get the gauntlet and palm done by tomorrow and can set off the fingers Saturday morning

If we get `undies-manual` underway and complete `tuck-guts` we might actually have caught up!

closed 2015-08-23 by cheapjack