cheapjack/buildyourown #77: Get the correct PLA stock for FACT reprints

cheapjack, 2015-08-18:
### FACT PLA stock

Silver 2 full spools
Grey 1 full spool 
Clear 1 full spool
Green 70% full spool
Cherry Red 1 half spool
Yellow full spool

**Colours required for reprints:**
`coil-mars` pink palm / light blue gauntlet
`beds-troy` silver palm / purple/darkblue gauntlet
`tuck-guts` red palm / yellow gauntlet
'gin-trite` dark blue / yellow

**Needs**
 * light blue
 * dark blue
 * pink ( I thought we had 2 pinks at FACT I delivered one to DoES but other one gone missing)

cheapjack, 2015-08-18:
Stock ok  use dark blue for all blues, updated reprint folder sent to Mark Murphy today

closed 2015-08-18 by cheapjack

LesleyTaker, 2015-08-18:
Sorry if I'm being dense. Just wanted to check you don't need us to order
anything PLA wise?

-- 

Lesley Taker
Curatorial Co-ordinator
FACT (Foundation for Art and Creative Technology)
88 Wood Street
Liverpool, L1 4DQ

Please note, I am only in office until 3pm on a Friday.
tw: @lexy_taylor
t: + 44 (0)151 707 4419
f: + 44 (0)151 707 4445

http://www.fact.co.uk
Admin: +44 (0)151 707 4444
Bookings: +44 (0)871 902 5737

*FACT - Bringing people, art and *
*technology together since 2003*

--------------------------------------------------------------
--------------------------------------------------------------

*Build Your Own: Tools for Sharing4 June – 31 August*

Showcasing four new commissions that span different

models of creative practice, including art, craft, and

creative technology, *Build Your Own* celebrates a new

spirit of generosity through making from your local

street to global maker networks.

--------------------------------------------------------------
--------------------------------------------------------------

-- 
 

*The contents of this email and any attachments are intended solely for the 
original recipient(s). If you have received it in error please contact the 
sender immediately by returning the e-mail or by telephoning a number 
contained in the body of the e-mail, then please delete the e-mail 
immediately without disclosing its contents. No responsibility is accepted 
for loss or damage arising from viruses or changes made to this message 
after it was sent. The views contained in this email are those of the 
author and not necessarily those of the authors employer or service 
provider.*

cheapjack, 2015-08-18:
It's OK you have enough stock to print the reprints this issue is now closed ta. :)