cheapjack/buildyourown #81: Generate a Raptor wing joint interface and place it in the repo for printing at FACT

cheapjack, 2015-08-25:
OpenSCAD fails to generate a scaled elbow interface for me in my out of date install of my old mac.

Can @amcewen plesae generate one at this scale and place in the repo? Then I'll deliver to FACT

I'm not sure what's best way of scaling? **eg** I've found if I take the generated `undies-manual`, and I scale it by 1.2 uniformly as below it 'looks' right in Openscad:
```
//joint interfacewingscalar = 1.2;

scale([wingscalar,wingscalar,wingscalar])translate([wingposx, wingposy, wingposz])import ("/Use
rs/Power5/buildyourown/cad/experiments/RaptorWingAdaptiveGripExperiment/elbow_joint_interface_r
ight.stl", convexity = 3);

```

I've added a `wing_arm` folder in the cad directory with a README.md [here](https://github.com/cheapjack/buildyourown/tree/master/cad/wing_arm/README.md

cheapjack, 2015-08-25:
Durr *think before you raise an issue* @cheapjack I've just realised I could scale by 1.2 in Cura so don't worry @amcewen if it's a good fit I'll ask you to generate others uniformly once we've tested it successfully with `undies-manual`

closed 2015-08-25 by cheapjack

amcewen, 2015-08-25:
If you want to scale it by the same amount as the hands, the formula that our OpenSCAD system uses is in cad/src/scaling.scad, but basically it's:

    (unaffected_measurement + 5)/55

So for undies-manual that'd be (59.21+5)/55 which gives a scaling factor of 1.167

cheapjack, 2015-08-26:
I printed it at 1.2 and it was way off so bit confused will have to leave it for another day