cheapjack/buildyourown #82: The X/Y belts are loose on both Ultimakers

cheapjack, 2015-08-26:
Someone passing through the exhibition suggested we get some [spring tensioners](http://www.ebay.co.uk/itm/Timing-Belt-Tensioner-Spring-Nickel-Plated-RepRap-CNC-3D-Printer-/151692235115) to tighten the belts without having to change them as they are looking super loose.

Need to look at them before sending on to Norwich

DefProc, 2015-09-03:
I've checked on the Ultimaker forums and google images of the black slider blocks, and the doesn't appear to be any adjustment for the "long" belts on the UM2. 

The long belt I saw was loose on the lower (attachment) side, which couldn't have a tensioner attached, but the end pulley can be loosened to allow redistribution of the slack, and allow addition of a tensioner on the other side. 

Metal spring tensioners are likely to be the better choice, but I did find a couple of printable ones too: 

 * http://www.thingiverse.com/thing:19892
 * http://www.thingiverse.com/thing:118917

Where are the Ultimakers now?

cheapjack, 2015-09-04:
They are currently at FACT @muff7 printed the tensioners out and they seem to help so that's great ta :+1:

closed 2015-09-10 by DefProc