cheapjack/buildyourown #85: Complete remaining hand prints under scope of Exhibition at DoES

cheapjack, 2015-09-04:
I've had an audit of what still needs printing before we pass onto the interest group and look at wing printing. We'd hoped to have been printing wings at this point but the re-print issue delayed us.

**Needs to start printing:**
 * `eddy-troops` pink palm dark blue extremity
 * `tuck-guts` red palm yellow extremity

**In progress see measuring workshop file for colour ref**

 * `ruing-beds` Needs 3 fingertips & 3 metatarsels from plates 1 & 2
 * `gin-trite` Needs plates 2 fingers from 1, 2, 3, 4, 5
 * `beds-troy` needs plates 1, 2, 4, 5 & Post. gushes-eddy can be posted as is with proviso, and tensioner pins have been printed
 * `grows-abort` Needs plates 3, 4, 5
 * `undies-manual` Needs wing & new gauntlet

**Complete**
 * `coil-Mars`
 * `gin-timers`
 * `tuck-choosy`
 * `manual-guts`
 * `gasp-law`
 * `month-reared`

cheapjack, 2015-11-05:
This is current status from the Measuring Workshop document:

**In progress Wings**

 * `troy-vane`

**Complete Raptor Wings**
 * `paddy-robust`

**Complete Raptor Hands**
 * `ruing-beds`
 * `coil-Mars`
 * `eddy-troops`
 * `gin-timers`
 * `tuck-choosy`
 * `tuck-guts`
 * `manual-guts`
 * `gasp-law`
 * `month-reared`
 * `tuck-guts`
 * `gin-trite` 
 * `beds-troy`
 * `peewee-undies`
 * `grows-abort` 
 * `undies-manual`

cheapjack, 2015-11-05:
Closing, referred to in BuildChecklist.md passed onto Prosthetic interest group

closed 2015-11-05 by cheapjack

amcewen, 2015-11-05:
Pah, if you'd left it another hour I'd have been able to mark this closed because all the printing is complete.  Just taken the final print for troy-vane off the printer :-)

cheapjack, 2015-11-05:
:) thanks for taking charge of remaining queue and wing scaling Ive not had chance to do much on that :)