cheapjack/buildyourown #90: Make some links to guidelines around the Experimental Raptor Wing

cheapjack, 2015-10-26:
Link to some guides to assembling and eventually re-sizing the raptor wing as part of the whole 'system'

cheapjack, 2015-10-26:
This is a rough guide I sent to Ana in Norwich which I can re-visit and turn into a .md guide

## Assembling a raptor wing

[image]

The Raptor wing is a work in progress from the e-NABLE forum below 

http://forums.enablingthefuture.org/viewtopic.php?t=53

We had to work out how it would assemble from this forum and had to play with a mix of screw lengths and widths etc. to put it together. 

It does help to see one assembled here for the left affected arm see above.

Key bit is working out how the actuator thread threads through the assembled parts and how the whipple tree works; it adaptively grips around the shape of an object so it grips in a different way.

https://github.com/cheapjack/buildyourown/blob/master/cad/src/RaptorWing/whippletree.stl

(see forum, its the bit between tensioner box and knuckle channels that turns the single actuator thread into a single one) 

I think we found that the whippletree needs to sit inside the tensioner 'box' (the tensioner with pins is replaced with a box that just makes a space for the whipple tree and thumb actuator thread); you can see the tensioner/screw endpoints are 'replaced' and moved into the cuff.

https://github.com/cheapjack/buildyourown/blob/master/cad/src/RaptorWing/box_left.stl

Then gauntlet connects to the elbow joint interface:

https://github.com/cheapjack/buildyourown/blob/master/cad/src/RaptorWing/elbow_joint_interface_left.stl

Threading wise 2 threads (one from whipple tree and t'other from thumb (green & orange in my image attached) go through the channel in the elbow joint interface and onward through the elbow joint ( https://github.com/cheapjack/buildyourown/blob/master/cad/src/RaptorWing/elbow_joint.stl
) then up into the cuff where there are 2 tensioner pins replacing the Raptor Hand tensioner block in the Raptor Reloaded Hand

Refer to the [images](http://forums.enablingthefuture.org/viewtopic.php?t=53) on the forum does help

This was all we had to go on the rest is trying it out Im afraid! It worked pretty well for one family. My enquiry in the forum about different files that would be better to edit and tweak in cad is on their radar so imagine Nick will get on to it when he can and maybe someone can look at that if necessary

Hope that helps for now.

amcewen, 2015-11-03:
In https://github.com/cheapjack/buildyourown/commit/f2880d773152709cd49673dfdb4de38a85b490bc I added the scale_wing script, which runs the same way as the scale_hand one but scales and sorts the files for the Raptor Wing